mu-editor/mu #11: Add flash capability

ntoll, 2015-12-08:
As a user, when I click "Flash", upon first click in the current editor session, I should be asked to identify where, on the file system, the micro:bit is located. Once selected, a .hex file is generated and copied to that location - thus flashing the device. On subsequent clicks the editor remembers the location of the device and misses out the selection step. If a non-micro:bit location is selected, the resulting `script_name.hex` file is copied to that location.

ntoll, 2015-12-14:
I'm going to start this.

funkyHat, 2015-12-16:
We're looking at this together.

We need to:
 - [ ] create the hex file
 - [ ] find device on the filesystem
 - [ ] copy the resulting hex file into the filesystem location

ntoll, 2015-12-17:
Work is ongoing here: https://github.com/ntoll/uflash PR's welcome. ;-)

ntoll, 2015-12-17:
(I had to change the package from `gordon` -> `uflash` because `gordon` was already registered on PyPi). :-/

ntoll, 2015-12-17:
@funkyHat OK... lots of work done late afternoon / early evening on uflash (see link above). Blocked by lack of automatic finding of device. ;-)

funkyHat, 2015-12-17:
Aha! Got this working for posix & nt, see https://github.com/funkyHat/uflash/tree/master/spikes

Should have looked here before making the commits on my fork... I'll shuffle stuff into uflash.py :)

ntoll, 2016-01-14:
Fixed by 70d7bde1b2e62ede8030cb7aa404913aee452da4

closed 2016-01-14 by ntoll