mu-editor/mu #2: Work out how to build static binary executables for Windows, OSX, Linux and RaspberryPi

ntoll, 2015-12-08:
Perhaps using this..?

https://pypi.python.org/pypi/pyqtdeploy/1.1

or even this:

http://www.py2exe.org/

- [ ] Windows
- [ ] OS X
- [ ] Linux
- [ ] Raspberry Pi

carlosperate, 2015-12-08:
For Py2exe you need to include with you application DLLs from visual studio, meaning you need to have the rights for redistribution (which I assume, but I am not entirely sure, probably means a license to the visual studio version that includes them). At this point I do not know if the newer free visual studio versions would include these dlls (only one is required, but 3 have to be packed together), all I can say is that last time I configured py2exe, before the VS community editions were released, it was a pain having to find and reinstall my old paid copy of Visual studio 2008 just to copy these files over, as I could not find them in the 2010 version. Things might have changed, but I though I should mention it in order to keep that in mind.

Pyinstaller should be able to work on all Win, Mac and Linux (in ARM I think you might need to build the pyinstaller bootloader locally, but it should still be doable), so this could be an option. There seems to be hooks for PyQt5 (https://github.com/pyinstaller/pyinstaller/tree/develop/PyInstaller/hooks), so hopefully it should work without fiddling too much about it.

I haven't used pyqtdeploy, but being a specialised tool it might be even easier to get it up and running, I guess we'll have to give it a go to see if its a good option.

ntoll, 2015-12-08:
@carlosperate this is very useful. Thank you!

I don't own a Windows machine any more. Fancy trying to get the very limited proof-of-concept we have so far into some sort of a state where we can create a static stand-alone binary exe..? I presume, once this is set up and configured it *ought* to be simply a case of re-running a command to re-build..?

Quite understand if the answer is no...

:-)

carlosperate, 2015-12-08:
I've got a quite few things on my todo-list for the week, so I won't be able to test pyqtdeploy, as I've never used it and I wouldn't have enough time to dedicate to it. Nonetheless, I am happy to give it a go with Pyinstaller or py2exe tonight. I assume pyinstaller might be preferred due to cross-platform compatibility, so I'll probably try that first.

ntoll, 2015-12-08:
Cool... I'll try pyqtdeploy :-)

carlosperate, 2015-12-09:
So, I manage to get something building in AppVeyor. Last night I spent way too long battling, and losing, with virtualenvs (first with Python 3.5, had to revert to 3.4, and then with PyQt5).

If somebody wants to test it, executables can be found in the link below, id' try to download the one with the latest time-stamp, but if I do break something, the latest working version at the moment of writing is "mu-2015-12-09_19_45_31.exe".

http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/index.html?prefix=microbit/windows/

For some reason none of the buttons do anything (not even load a new script), and the `win_package` should be synchronised with the current upstream master. I had the same problem when running the non-packaged version on my Windows machine, but at the time I assumed they were just not implemented. What is meant to be working right now?

carlosperate, 2015-12-09:
Nevermind, [mu-2015-12-09_20_12_01.exe](https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/windows/mu-2015-12-09_20_12_01.exe) works, not quite sure what happened to the previous one.

inglesp, 2015-12-13:
In order to track this better, I've created an issue for each of the four platforms in question (#25, #26, #27, #28).

inglesp, 2015-12-13:
@carlosperate, am I right in thinking that you've got a working build process for Windows?  If so, could you document this on #25?

closed 2015-12-13 by inglesp

carlosperate, 2015-12-13:
Got a working version for windows and linux in a branch that also tries to build for all platforms. I will submit a WIP pull request for visibility, to track progress, and hopefully to receive comments and start the review process.

I'll link the individual issues once I submit the PR.

inglesp, 2015-12-13:
Nice one @carlosperate!

carlosperate, 2015-12-13:
submitted: https://github.com/ntoll/mu/pull/29