mu-editor/mu #19: Editor doesn't open on launch in Windows 10

carlosperate, 2015-12-09:
Tested on Windows 10 x64.
When the application is launched (just `python run.py`) the editor is not opened, just a white box appears on its place. The editor does open fine when the load button is used.

ntoll, 2015-12-10:
Hi, apologies... that's me just getting this done mid-file related CRUD. Will fix presently.

NDevox, 2016-01-09:
Just tested this in win10 x64.

Python 3.4.4

PyQt5 binary downloaded from here: https://riverbankcomputing.com/software/pyqt/download5

Works fine for me @carlosperate is this still broken for you or can the ticket be closed?

ntoll, 2016-01-10:
I think this can be closed.

carlosperate, 2016-01-10:
Running this in in Win10x64, python 3.4.2, PyQt 5.4.2 and this is what I currently see when launching the application:

[image]

I assume an empty editor should be opened on launch, as it did in the very early versions?

ntoll, 2016-01-10:
No. :-( Is this the latest build based on the most recent merge into master?

It should attempt to open the scripts that were active when last used, or start with an untitled script. There should be a tab with code in it in any case.

carlosperate, 2016-01-10:
The screenshot has been taken with the latest master, including the latest merge, but this has been the case for me since first reported here.
This is also the case when testing the builds created in the appveyor servers.

ntoll, 2016-01-10:
OK... then it's definitely wrong. I suspect it's related to #49 since the code to create the untitled tab won't be reached because of the error/bug in #49.

carlosperate, 2016-01-10:
I doubt it is at all related to #49, I got around that by manually creating the user directory. And as I mentioned in my previous message, I've experienced this issue since the 9th of December, so it was caused by a commit previous to that moment.

ntoll, 2016-01-10:
Ah... OK... this is all guesswork on my part. I'll test/check/fix in the morning.

closed 2016-01-11 by ntoll