mu-editor/mu #26: Work out how to build static binary executables for OSX

inglesp, 2015-12-13:


inglesp, 2016-01-03:
I'm running into problems with getting PyInstaller to build something on OSX.  The build process apparently completes successfully, but when I come to run the application, I get exception `ImportError: No module named 'PyQt5.Qsci'`.

@carlosperate or anybody else, do you have any ideas for how I can go about debugging this?

carlosperate, 2016-01-03:
I assumed instaling Qsci was a different step than what is currently on the [install_osx.sh](https://github.com/ntoll/mu/blob/master/package/install_osx.sh) file, mostly because of the import error you mention, and because I had to `sudo apt-get install python3-pyqt5.qsci` in ubuntu to get it to run.

Is your build environment able to launch the python program without throwing he same import error? If so, there is a small chance that PyInstaller is not figuring out the import requirements properly, and it can be added to the spec file as a "hidden import", however there is a clear dependency in the [editor.py](https://github.com/ntoll/mu/blob/74b6ac9a075363f5d3f8df89fc270f50fdd8544c/mu/editor.py#L11) file, so I'd be surprised if that would solve it.

There is also a chance that it might be a fixed bug in PyInstaller. You check that by downloading the source code directly from github and run [that pyinstaller.py file](https://github.com/pyinstaller/pyinstaller/blob/develop/pyinstaller.py) instead of the `pyinstaller` command.

inglesp, 2016-01-04:
Thanks @carlosperate for your input.  I *am* able to launch the Python program without any problems -- I probably should've made that clearer yesterday!

`build/pyinstaller/warnpyinstaller.txt` contains a bunch of warnings about missing modules, which includes:

```
missing module named 'PyQt5.Qsci'.QsciLexerPython - imported by 'PyQt5.Qsci', mu.editor
missing module named 'PyQt5.Qsci'.QsciScintilla - imported by 'PyQt5.Qsci', mu.editor
missing module named 'PyQt5.Qsci' - imported by mu.editor
```

Further digging shows that PyInstaller is trying to load the `QSci` module from the wrong directory -- it's looking in `/usr/local/Cellar/pyqt5/5.5.1/lib/python3.5/site-packages/PyQt5`, but it should be looking in `/usr/local/lib/python3.5/site-packages/PyQt5`.

The first path is where homebrew puts the packages before symlinking them to the second path. I'll need to spend some more time getting cosy with `pdb` to work out why it's looking in the second path.

inglesp, 2016-01-05:
Ooh ooh ooh, I've commented out two lines in PyInstaller, and I've got something to work!  See https://github.com/pyinstaller/pyinstaller/issues/1750 for details.

carlosperate, 2016-01-05:
That's great progress @inglesp! Let's see how the bug report goes, if it gets fixed on their git repository we can always pull that on the travis ci installation script. If not, we could have a bit of a hacky-temporary-solution to clone and apply a patch before using PyInstaller.

closed 2016-01-14 by ntoll