mu-editor/mu #28: Work out how to build static binary executables for Raspberry Pi

inglesp, 2015-12-13:


carlosperate, 2016-01-10:
Just to quickly mention that I just tested the latest version in raspberry pi model b (so the old ARM6 version), and it seems to work fine, although for obvious reasons I wasn't able to test the flash and repl.

I've got the following text on the console, but I would assume there really isn't much we can do about them:

```
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:${ORIGIN}/dri:/usr/lib/dri)
Cant find EGLConfig, returning null config
Unable to find an X11 visual which matches EGL config 0
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
Cant find EGLConfig, returning null config
```

carlosperate, 2016-01-10:
Btw, do we want to create static binary files for the pi? since we have to target 2 different platforms (ARM6 and ARM7), and all the requirements are easily installable with apt packages, wouldn't it be better to work on a debian package that could be installer from the command line?

ntoll, 2016-01-11:
@carlosperate I agree with you... I believe we should just create a .deb package that can be `apt-get install`-ed I think we should ping @bennuttall to see if he can help give pointers for how to do this.

As far as I can tell, we'd "only" need to package the content of this repos and make sure it had a dependency on the appropriate Python / Qt5 packages (notice the air-quotes - I've never packaged for Debian before so would have to read the docs to discover how much of an effort it is).

bennuttall, 2016-01-11:
Hey sorry I didn't get to test this over Christmas, busy time. Anyway, I've tested the executable on my Ubuntu laptop and it seems to work. However, no luck getting it to work on a Pi.

I've installed the dependencies and downloaded the latest Linux build from [here](http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/index.html?prefix=microbit/linux/). I'm running `chmod +x` on it and executing (the same way as on my laptop) but I just get:

```
-bash: ./mu-2016-01-11_08_01_50: cannot execute binary file: Exec format error
```

Any suggestions?

Once we've got it working I'll pass it on to our engineers to work out a way of getting it into Raspbian. A working deb, or ideally a package in the Debian repository (get it into Ubuntu too) would be ideal. We'll do some evdev magic to get the editor to boot as soon as a microbit is connected to a Pi.

ntoll, 2016-01-11:
So, the linux binary will target x86-64 rather than the ARM architecture of the Pi. As far as I can tell, you only need the appropriate QT libraries and Scintilla installed and then just run the project with Python 3. I'll "borrow" Sam's RPi v1 B+ this evening, have a play and report how I get on. AFAICT "it should just work" (famous last words). 

The only thing that'd need packaging is the content of this repos as I describe above.

carlosperate, 2016-01-11:
Hi @bennuttall , to get it running I've basically just followed the installation instruction from the second step in the [package README file](https://github.com/ntoll/mu/tree/master/package) and it worked fine on the model 1 B:

```
sudo apt-get install python3-pyqt5
sudo apt-get install python3-pyqt5.qsci
sudo apt-get install python3-pyqt5.qtserialport
```

carlosperate, 2016-01-11:
A general question about deb packages: I assume any missing dependencies are installed from the internet using `apt`, is there any option to pack them inside the deb file? As the one possible advantage of using the built executable would be easy offline installation.

ntoll, 2016-01-11:
They get automagically installed via the network.

bennuttall, 2016-01-11:
Ok I'll try that.

Debian policy not not allow bundling of dependencies - it's all about shared libraries. However, I think this has been done for a few Raspberry Pi packages.

We'll pre-install the microbit software and dependencies in Raspbian so that anyone with an up-to-date image will have it automatically. And for anyone else, it'll just be a single `apt-get install`.

bennuttall, 2016-01-11:
Ok tried building but got a fatal error:

```
Fatal error: PyInstaller does not include a pre-compiled bootloader for your
platform. See <http://pythonhosted.org/PyInstaller/#building-the-bootloader>
for more details and instructions how to build the bootloader.
```

ntoll, 2016-01-11:
@bennuttall yeah... we should go the .deb route. I've never built a .deb... looks like I'll have a go this evening. :-)

carlosperate, 2016-01-11:
@bennuttall PyInstaller is only required to build the application in a self contained executable, so it is not needed to run mu.

ntoll, 2016-01-11:
OK... I can confirm that installing the three packages @carlosperate mentions and typing:

```
python3 run.py
```

...in the root of the repository starts the editor and it all works right out the box. Yay! :-)

I'm going to close this issue and create a new one for packaging this work as a .deb package for Debian et al. This should only be done once we have a relatively stable release (it's still currently a work in progress).

closed 2016-01-11 by ntoll

bennuttall, 2016-01-12:
Ok that works for me too. Great! However it's not connecting to my microbit. Need to remember how to flash it...

ntoll, 2016-01-12:
Click the "Flash" button with a blank Python script. ;-)

bennuttall, 2016-01-12:
Ah ok that wasn't working earlier. Working now! \o/

ntoll, 2016-01-12:
@bennuttall there's also this: https://github.com/ntoll/uflash which should also be packaged (once I've merged @funkyhat's work on de-hexifying). It provides a simple command line tool for flashing scripts onto the device. See: http://uflash.readthedocs.org/en/latest/ :-)

carlosperate, 2016-02-11:
Until we get the deb package sorted I was playing with pyinstaller on the raspberry pi and got this built version up and running:
http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/raspberry_pi/

It's obviously not something meant to be maintained (although it wouldn't be too complicated to add a github hook and automatically kick a build on raspi the same way the CI servers do), as the deb solution will be much better to ensure users can easily update install and update mu. Nevertheless, I though maybe having a simple to use binary for the new wave teaches could be useful.

Is it worth having these binaries as well in the meantime? Or should we not confuse users by having two different delivery methods? thoughts?

carlosperate, 2016-02-11:
Btw, it built it on ARMv6 (Pi 1) and works on ARMv7 as well (tested only on a UDOO Neo board, not Pi 2, so I would welcome if somebody could test it there).
The reason I've named the folder `raspberr_pi` instead of `linux_arm` is because I've not used the LSB binary option since I couldn't find the LSB build tools for ARM, so there might be compatibility issues depending on the linux distribution. Debian/Ubuntu seem to be working fine though.

ntoll, 2016-02-12:
@carlosperate stonking work matey. :-)

carlosperate, 2016-03-18:
Just to let you know that we have a new tiny automated CI server for the raspberry pi builds (even though they might not be required for much longer), still need to configure some extra stuff, but the basic functionality is up and running.

[image]