mu-editor/mu #30: Decouple view code from business logic

alistair-broomhead, 2015-12-14:
This should make other tickets easier to work on, especially #4 - adding tests

alistair-broomhead, 2015-12-20:
On Wednesday we decided that 1. is worth a try, as is 2., and for 3. we're going with MVC for now. I then promptly tries to do everything at once and ended up with a mess,

Today I've gone back to the drawing board, and I'm trying to do one thing at a time on https://github.com/alistair-broomhead/mu/tree/feature/30-decouple-view-code-from-business-logic, starting with 3. - I'm currently re-factoring 62e193f0636f2678a3c68af16fce821e1a01cc7c to MVC, then I'll add in what's been done since then to get an MVC version of the code-base.

After this I shall attempt to make the models more functional, and then look at porting the view code to QML.

inglesp, 2015-12-21:
Hi @alistair-broomhead -- when do you think you might have something ready to be reviewed and merged?  I ask, because it's clear you're making some quite big changes on your branch, and the more changes there are on master, the more work you'll have to do to keep up.  As such, if you were going to produce a reviewable pull request by, say, the end of tomorrow, we could hold off merging other things, but we obviously don't want to be doing that forever.  What do you think?

alistair-broomhead, 2015-12-21:
@inglesp It's possible, but I wouldn't say definitely. There are going to be merge conflicts, someone is going to solve them for each set of changes, and it probably makes sense for those all to be resolved by one person, to avoid anything being missed. What else is likely to need merging?

ntoll, 2015-12-21:
@alistair-broomhead if you get the PR ready, I'll review and resolve. Would like to get this change in ASAP so I (we) can hack on the app over Xmas hols. ;-)

ntoll, 2016-01-14:
The initial work for this has been completed.

closed 2016-01-14 by ntoll