mu-editor/mu #31: Find teachers and other interested people to act as crash test dummies!

ntoll, 2015-12-14:
We should release early and release often. The sooner we get this project in front of teachers and other potential users the sooner we can get actionable feedback.

We should gather together a cohort of willing crash-test dummies who are willing to feel the pain of pre-Alpha software.

This groups must include:

 - [ ] Teachers.
 - [ ] Non-technical parents.
 - [ ] School technicians and/or sys admins.
 - [ ] Classroom assistants.
 - [ ] Professional programmers who are not authoring this project.

We should include clear first steps instructions and a well defined mechanism for creating feedback and interacting with the project.

'nuff said. :-)

jonwitts, 2016-06-07:
Happy to be one of your crash test dummies!

ZanderBrown, 2016-06-07:
Not sure I want to be a 'crash test dummy' as such but ill happy test mu

ntoll, 2016-06-08:
Great stuff... there will be a new release soon and I'll ping you when we get close. ;-)

ntoll, 2018-06-04:
"Fixed" by modus operandi explained here: https://mu.readthedocs.io/en/latest/user-experience.html

closed 2018-06-04 by ntoll