mu-editor/mu #32: Onboarding for new users...

ntoll, 2015-12-14:
In light of the results from #31 we should ensure the "on boarding" experience is, at the very least, appropriate for an 11yo kid, their teacher and / or parent. Not sure how useful something like: https://www.useronboard.com/onboarding-teardowns/ would be for ideas in this respect. Also, for design work we could ping @hawkz for instant awesomeness in the design department.

Avaren, 2016-03-17:
So there have been quite a few issues citing usability/teaching features. Another thing that has been mentioned a few times is using game logic.

Here is a quick mock up, what does everybody think?
[image]

ntoll, 2018-06-04:
While "quests" is something I really like, I think all the work happening on the new website will also address this in the immediate term. If quests will be a thing, then quests should have its own ticket post 1.0.

closed 2018-06-04 by ntoll