mu-editor/mu #37: Add keyboard shortcuts

NDevox, 2015-12-17:
Purely for ease of use. 

Basic ideas include new tab, open, save, zoom in and zoom out.

NDevox, 2016-01-15:
Realistically I don't think there are many more shortcuts that need to be added.

Can always re-open, or open a new ticket, if we think of any more.

ntoll, 2016-01-15:
Hi @NDevox, I've been chatting with some visually impaired developers about making "mu" accessible for all. The bottom line is that we'll need a shortcut for at least *all* the menu items.

What we need to work out are the conventions for such short-cuts. I'll give them a ping and see what they say. I'll re-open the issue just so we have somewhere to discuss this further as the information comes in.

Apologies for not making this clear sooner.

NDevox, 2016-01-15:
Interesting!

I hadn't thought of it that way. Let me know what they say.

My initial thoughts are we could use Ctrl+enter for REPL. And, if it's
possible (I believe it is) we could offer multiple shortcuts. Maybe
standard letter based ones. And then something more pattern based. Such as
using Ctrl+1-9.

Let us know what they say :-).

carlsmith, 2016-07-13:
There are keybindings for increasing and decreasing a blocks indentation, but there's apparently no way to move a block up and down. I know the aim is to keep things simple, but being able to move blocks vertically is really helpful.

I couldn't actually find any documentation on keybindings.

ladyada, 2017-08-15:
ditto for this, Control-S doesn't save, Control-N doesn't make a new tab, Control+ and Control- don't zoom in/out and its making my brain hurt :D

ntoll, 2017-08-15:
Hi Folks,

I just had a quick look at this, it turns out that the code that set hot-keys wasn't configuring things in the right way. I've hacked a preliminary fix that works and I'll merge the change when I get home from holidays!

My aim is to have keybindings for all the menu items and it should be possible to expose Scintilla's keyboard shortcuts in the same way (see: http://www.scintilla.org/ScintillaDoc.html#KeyboardCommands). In any case, new, load, save, zoom-in and out all work in my hacked together test.

Thanks for the kick!

N.

ntoll, 2018-06-04:
All the menu items have keyboard shortcuts and the new website will even have a tutorial about it. Closing this issue since "shortcuts" is a baked in expectation for Mu features.

closed 2018-06-04 by ntoll