mu-editor/mu #39: Document the install process for PyQt5

NDevox, 2015-12-17:
I'm on Ubuntu and it wasn't the clearest setup (although doesn't take long).

My steps were:

sudo apt-get install qt5-default
sudo apt-get install python3-pyqt5
sudo apt-get install python3-pyqt5.qsci
sudo apt-get install python3-pyqt5.qtserialport

Let me know if anything is missing.

Not entirely sure where this kind of information should be put as well.

ntoll, 2015-12-17:
Hi @NDevox, this is great... as you probably figured out, this is only around a week old so all this if first steps. I'll add some obvious docs for this sort of thing. Thanks!

NDevox, 2015-12-17:
No worries, I know it's a young project just adding in my thoughts :-)

carlosperate, 2015-12-17:
As part of the self-executable packaging PR (https://github.com/ntoll/mu/pull/29/files), there are two shell scripts for installing the dependencies on the Ubuntu and OS X servers (also a short README with the same info). Of course is not quite as good as having proper documentation, but until then I am mentioning here as it could be a useful place to look into.

NDevox, 2016-01-15:
If we get PyQt5 in under the dependencies in the setup.py then we might be able to discount this issue.

Would it be worth moving the other dependencies in as well?

NDevox, 2016-01-15:
Yeah looking this up apparently PyQt5 can't be installed by pip (as I should really know by now). So disregard the above. http://stackoverflow.com/questions/12607241/python-pip-no-distributions-at-all-found-for-an-existing-package/13662333#13662333

Is there a way to package it in with everything else?

carlosperate, 2016-11-18:
PyQt can now be installed using pip, this is discussed in https://github.com/mu-editor/mu/issues/120.

As far as this PR is concerned I'd say it is covered by https://github.com/mu-editor/mu#development and this could be closed?

closed 2016-11-19 by ntoll