mu-editor/mu #40: Zoom_in/zoom_out throw AttributeError for repl

NDevox, 2015-12-17:
running `zoom_in` and `zoom_out` throws an `AttributeError` for repl:

    Traceback (most recent call last):
      File "/home/milo/Projects/mu/mu/editor.py", line 330, in zoom_in
        self.repl.zoomIn(2)
    AttributeError: 'function' object has no attribute 'zoomIn'

I'd suggest wrapping the relevant section of code (`self.repl.zoomOut(2)`) in an `if hasattr` statement, as is done for the tabs.

Is there a reason why this hasn't been done?

ntoll, 2015-12-17:
It was done quickly, it was on my todo to fix (noticed it last night at the London Python Project Night).

NDevox, 2015-12-17:
Fair enough!

I'll leave it for now as already have an open pull and might cause a conflict if I put in another one.

Can put in the above quick fix if you'd like though.

ntoll, 2015-12-17:
Please do.

NDevox, 2015-12-17:
Added into current open pull #38

NDevox, 2015-12-17:
Closed as above merged.

closed 2015-12-17 by NDevox