mu-editor/mu #41: Ensure REPL works cross platform.

ntoll, 2015-12-19:
Currently the REPL only works in Linux. This needs to change so that it works in:

- [ ] Windows 7/8/10
- [ ] OS X
- [ ] Raspberry Pi

Given the cross platform nature of Qt this (hopefully) shouldn't be an onerous task. The major sticking point will be upon device detection.

Unfortunately, Windows does not come with serial/COM drivers for the micro:bit installed by default (happily, Linux and OSX do).

If you're on Windows, in order to make this work you should use the mbedWinSerial_16466.exe file (found here: https://developer.mbed.org/media/downloads/drivers/mbedWinSerial_16466.exe) to install the correct drivers. You MUST have your micro:bit plugged into your computer when you run this command.

For more information and the latest versions of this driver please visit:

https://developer.mbed.org/handbook/Windows-serial-configuration

inglesp, 2015-12-19:
Why do we think it doesn't work on OS X?  It's worked for me!

ntoll, 2015-12-19:
@inglesp it just needed someone to check it! ;-) Thanks.

ntoll, 2016-01-11:
Closing because all three target platforms have been tested and shown to work.

closed 2016-01-11 by ntoll