mu-editor/mu #45: Appveyor builds fail due to AWS secret key error

ntoll, 2015-12-27:
The Windows build appears to work. This is awesome, apart from one fly in the ointment. At the very end the process errors with the following message:

```
Value cannot be null.
Parameter name: awsSecretAccessKey
```

While I can see the asset is created, the actual build is marked as failed. 

Is there any way to fix this?

carlosperate, 2016-01-03:
We currently have a working AppVeyor project running at: https://ci.appveyor.com/project/carlosperate/mu , which deploys the built executables to http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/index.html?prefix=microbit/windows/

However, the wrong badge is attempted to be displayed in the [package/READM.md](https://github.com/ntoll/mu/blob/master/package/README.md#download-the-mu-executable--) file, so we still need to update that.

carlosperate, 2017-09-27:
This is not an issue anymore.

closed 2017-09-27 by carlosperate