mu-editor/mu #49: Error creating user data under windows

carlosperate, 2016-01-10:
I am getting the following error when running mu under Win10x64.

```
(mu3) E:\workspaces\git\mu>py -3 run.py
Qt: Untested Windows version 10.0 detected!
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "E:\workspaces\git\mu\mu\app.py", line 24, in <module>
    from mu.editor import Editor
  File "E:\workspaces\git\mu\mu\editor.py", line 35, in <module>
    os.mkdir(DATA_DIR)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\Carlos\\AppData\\Local\\python\\mu'
```

The folders `python\mu` are not present in the `C:\Users\Carlos\AppData\Local` directory, but the application catches the user app data local directory correctly.

ntoll, 2016-01-10:
My guess is that I should be using `os.mkdirs` not `os.mkdir` viz: https://docs.python.org/2/library/os.html#os.makedirs

Will test/fix in the morning.

carlosperate, 2016-01-10:
You were completely right @ntoll, tested and PR submitted.

closed 2016-01-11 by ntoll