mu-editor/mu #57: Save As.../Rename

NDevox, 2016-01-11:
Just wondering along the lines of expected usability.

Would it be worth putting in a `save as...` method? If someone accidentally named their file something other than expected it would currently prove a minor headache to change the name. Whereas `save as...` would just be a minor variant of the current `save` function.

I have two thoughts on this process if we were to implement it:

1) I don't think we should add an extra button to the toolbar. But maybe have a menu which could repeat a few of the buttons and add extra options (e.g. `save as...`)

2) Something fun to code could be a tab renaming option, where double clicking the tab name allows you to rename it.

1 I think could be something very useful.

2 is relatively hidden and is just a small UX enhancement which may go unnoticed, so not really necessary, but then the little things always make the difference.

Any thoughts?

ntoll, 2016-01-14:
I like option 2. Perhaps if you double click name in the tab it becomes editable and pressing return does the equivalent of a `mv old.py new.py`. Something for a 1.1 release perhaps..?

NDevox, 2016-01-14:
Depends on when the 1.0 release is scheduled for!

I know it is soon though so it's a likely 1.1 enhancement.

Option 2 is definitely my favourite. What about number 1 though? I don't see them as mutually exclusive and feel that 1 is pretty valuable.

cymplecy, 2016-02-05:
I like option 2-i even tried to see if that worked when I couldn't find a save as function.  They only concern is that save as is standard method of working but changing tab name isn't but I still like it

NDevox, 2016-02-05:
My opinion is to do both. But happy to wait for V 1.0 to be out before implementing. When I have some time I'll set both up and put in a PR.

alistair-broomhead, 2016-02-05:
Perhaps worth thinking about for post V1.0, is some sort of onboarding tutorial, whether it's a) a guided tour through the application, b) a "tip of the day" feature, or c) just a set of help pages that guides you through everything from a beginner's eye view. Again these aren't mutually exclusive. c is possibly the easiest, a is possibly the best for discovery, and b isn't a bad half-way house ;)

NDevox, 2016-02-05:
@alistair-broomhead In my mind all 3 are a pretty good idea with c being necessary, a being really helpful and b being quite fun.

Although I feel the number of hints we could give would be quite limited.

Probably worth a separate issue as well so we don't get it lost in this one.

alistair-broomhead, 2016-02-09:
@NDevox I've added a new issue #81 for these ideas

ZanderBrown, 2018-03-23:
So double-clicking a tab now seems to act as Save As, does that resolve this and #72?

ntoll, 2018-03-23:
Yes

closed 2018-03-23 by ntoll