mu-editor/mu #58: Create a new .deb package so the editor is easy to install on Debian based systems.

ntoll, 2016-01-11:
We should package the "Mu" editor for Debian based systems. This will include Debian itself, Ubuntu and, of course, Raspbian.

The package should have three dependencies: `python3-pyqt5`, `python3-pyqt5.qsci` and `python3-pyqt5.qtserialport`.

We should only package the appropriate parts of this repository. The process for packaging and installing from such a package should be documented in the README here:

https://github.com/ntoll/mu/tree/master/package

ntoll, 2016-01-12:
@bennuttall this is the ticket for .deb-ifying the editor.

bennuttall, 2016-01-12:
Great - thanks

ntoll, 2016-01-12:
I'm going to go through this process to get the package into Debian:

https://www.debian.org/devel/wnpp/

asb, 2016-03-25:
See PR #102 for a start on this.

carlosperate, 2016-04-17:
Unless we want to publish the deb package somewhere else, looks like we can close this issue :)

ntoll, 2016-04-18:
Is there a link somewhere "upstream" (Debian?, Raspbian?) to which we can point for the package. It'll mean that this issue resolves by pointing at the solution. :-)

bennuttall, 2016-04-18:
[This list](http://archive.raspberrypi.org/debian/dists/jessie/main/binary-armhf/Packages) points at [this location](http://archive.raspberrypi.org/debian/pool/main/m/mu/) where the deb can be found.

bennuttall, 2016-04-18:
Also

```bash
sudo apt-get update
sudo apt-get install mu
```

rotacoder, 2016-07-29:
I've downloaded the mu binary for my Debian system but when I try to run it I get the error "cannot execute binary file". I assume this is because the binary is for a 64 bit system and I'm running 32 bit. Is this correct and if so will there be a 32 bit binary available in the future? Thanks.

carlosperate, 2016-07-29:
Looks like now that Mu has started using PyPI dependencies not available through the OS package manager we need to figure out a way to include these.
I'm living this link here for future reference, as it might be a good way to solve this: https://github.com/spotify/dh-virtualenv

@asb have you seen or used dh-virtualenv before?

jaustin, 2017-01-03:
@carlosperate is this actually fixed now? I notice there's a debian directory rules/control file etc?

carlosperate, 2017-01-04:
The dependency issues have been fixed (we have "temporary" fall back on the code due to package rename). The files in the debian directory are up to date (the changelog is in sync with the history file), so a deb file and associates files can be created by running the respective command line tools.
As a quick note, if we were to create a new deb file right now, it should be done on https://github.com/mu-editor/mu/commit/c2a0f23e1bd1c5245f877d9c6274dc87bda50b8f , as that's the last documented release. For a newer release we should update the changelog.

jaustin, 2017-01-04:
> be done on c2a0f23

Cool, thanks for the explanation - I wasn't actually planning on building
any packages yet, just more interested about how ready it was to happen -
Looks like there's not a package in upstream Debian at the moment?

gquintana, 2018-05-14:
To be able to install a mu .deb package on, at least Debian stable and Ubuntu LTS, some dependencies requirements should be downgraded:

```
python3-pyqt5       required:5.10 debian_stretch:5.7   ubuntu_xenial:5.5.1 ubuntu_bionic:5.10.1
python3-pyqt5.qsci  required:2.10 debian_stretch:2.9.3 ubuntu_xenial:2.9.1 ubuntu_bionic:2.10.2
```
This would also make installing mu with pip easier (no specific virtualenv required).

ntoll, 2018-06-04:
We're also stymied since there's been zero action on this (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892198) in upstream Debian. Any suggestions for how to move this forward?

gquintana, 2018-06-04:
@ntoll An "in between" solution could be to provide a .deb package and let users download it and install it using `dpkg -i`, at least until this package get published in Debian repos.

bennuttall, 2018-06-04:
I believe there's also a plan to have this package available in our PPA: https://launchpad.net/~rpi-distro/+archive/ubuntu/ppa

But yeah, a .deb on the site would be handy too!

ntoll, 2018-06-04:
@bennuttall if there's a plan, it'd be good if this were properly communicated. Open collaboration etc etc etc ... (I know I'm preaching to the choir here, just saying this here so it's logged for future reference). ;-)

bennuttall, 2018-06-05:
@XECDesign can you confirm it would be possible to add to our PPA once Mu is released?

XECDesign, 2018-06-05:
It's possible, but that will be decided once we have a release ready to go for Raspbian. I'd recommend focusing on getting the package into Debian and Ubuntu properly. That will help steer development in the right direction and avoid adding difficult dependencies.

ntoll, 2018-06-05:
Hey hey @XECDesign, the only difficult package that I can think of right now is referenced in this bug in Debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892198

Is there anything we can do to help make this happen..?

XECDesign, 2018-06-05:
I haven't been following the development since Peter's feedback, so I don't know how much of it has been addressed. I'm only responsible for getting the release into our distro of Raspbian.

ntoll, 2018-06-05:
Completely understand. Busy people are busy (aren't we all?). 

Can you suggest anyone upstream-Debian related who may be able to sit down with me (perhaps in London) so I can buy them food/drink in exchange for answering my questions and pointing out how best to unblock the various issues.

See: https://twitter.com/ntoll/status/1004000758352764929

bennuttall, 2018-06-05:
@ntoll Have you tried contacting Riverbank about the PyQtCharts package? I don't know if they are responsible or involved in the Debian packages for their software.

ntoll, 2018-06-05:
Honestly..? Right now I'm up to my eyes in all sorts of Mu related things which I'm trying hard to juggle, resolve and make good for a 1.0 release. I simply don't have the time to keep asking people (like Riverbank and/or Debian related folks) about yet more stuff they're only going to say they don't have time to do.

My hunch is if Riverbank were responsible for packaging PyQt and associated modules for Debian, it would have already been done. Furthermore, someone may even have replied to my bug report about the missing package.

I can't help but feel like I'm going round and round in circles here. :-(

XECDesign, 2018-06-05:
I'm not in that world, so I wouldn't be able to point to anybody. Peter might know of some Debian conventions or meet up groups which could be helpful.

ZanderBrown, 2018-06-05:
DebConf is in Taiwan I believe so probably not convenient...

ntoll, 2018-06-05:
Taiwan would be fun, if time consuming @ZanderBrown.

@XECDesign, completely understand... I'm just trying to effectively, politely and collaboratively unblock something that's been unresolved for far too long (and which if I could do myself, I would have done ages ago).

gquintana, 2018-06-05:
@skitt can you give some advice?

ukBaz, 2018-06-05:
No idea how up-to-date the following information is:
https://wiki.debian.org/LocalGroups#UK
https://blog.einval.com/

ZanderBrown, 2018-06-05:
Me & IRC often end badly & I'm really not in the Debian world but let's see if that gets us anywhere

skitt, 2018-06-05:
Nice, a Microbit-friendly editor... Tell you what, I’ll take care of PyQtChart; I’ll move the bug to the appropriate place (WNPP as @ntoll just found out on Twitter) and package it up. I can follow up on Mu itself too. (I’m a DD and I maintain a number of Python packages already.)

ntoll, 2018-06-05:
Hi @skitt -- first of all, THANK YOU. Secondly, I'm more than prepared to move mountains, do boring stuff etc... assuming the caveat that I'm a technical savvy Debian ignoramus (when it comes to packaging), so if there's anything you need me to do, don't hesitate to ask. Third, I assume this will go something like sid -> test -> stable (when testing is switched to stable). Finally, Just having PyQtChart available is a win, let alone having Mu packaged. Once again, thank you.

russel, 2018-06-05:
@skitt I was going to issue a ITP and actually try building the package prior to finding a mentor. I was looking for a suitable package to follow, preferably to create an 'all' package as I believe this package does not require any shared objects – but I may be wrong. If you have a Python package building Git repository for one of the other Python3 packages I'd be happy to chip in by trying to make a workable package for PyQtChart based on it.

ntoll, 2018-06-05:
@russel there are already Debian related packaging things in the repos, but they're a bit of a black box and the version numbers of dependencies in setup.py are often > than what's currently in Debian (testing).

skitt, 2018-06-05:
@russel oops, sorry, I didn’t mean to step on anybody’s toes. I’ve turned https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892198 into an ITP, if you want it you can take it, or we can work on it together (I’ve started on it but I’m still at the “listing build dependencies” stage). Given that this is a PyQt5 module, the best repository to start from is PyQt5’s in https://salsa.debian.org/python-team/modules/pyqt5. As @ntoll says you’ll need to work with the packages in unstable, the versions in Debian 9 are too old, and anyway packages always target unstable first.

russel, 2018-06-05:
@ntoll The debian directory relates to packaging mu itself, a precursor is getting the package python3-pyqt5.qtchart (and python3-pyqt5.qtchart-dbg) created and in Debian Sid. Once that is in place changes will be needed to the mu dependecies in the debian directory files.

My question to @skitt was whether he had a python3-??? package building repository I could use as a paradigm to create one for python3.pyqt5.pychart.

ntoll, 2018-06-05:
@russel aha... you're a :star: :-)

russel, 2018-06-05:
@skitt Toes not being trodden on at all. The goal is to get the package in as soon as possible. I was just trying to help out after @ntoll 's Twitter request.

The last time I did any packaging was in 2004 when one of my startup staff was deeply into putting stuff into Debian and got us to switch from RH to Debian – damn good choice. He was the DD, I was the person doing what I was told. :-)

I'm a total beginner to modern Debian Packaging, so collaboration with and working to the requirements of a DD sounds excellent. If you want me to act as your gopher on this, that works for me.

(I use Debian Sid and Fedora Rawhide, anything else is far too outdated. I guess I should also have an Arch set up. :-)

ntoll, 2018-06-05:
Just a FYI... lost in all the traffic. We aim to be releasing a 1.0 version of Mu very soon. It is *that* which will need to be the version ultimately packaged. I'm trying to front-load effort so "it just works" when the 1.0 balloon goes up, IYSWIM.

darrell24015, 2018-06-06:
When it gets to the point of testing, I’ll be happy to try out Debian, Ubuntu and Raspbian installs.

russel, 2018-06-07:
@skitt If you have a URL for the repository you have for "python3-pyqt5-qtchart" package I'll clone it and see if I can get some positive contributions in place.

plugwash, 2018-06-07:
I am a Debian Developer.

I looked at packaging mu a while back, the two main missing dependencies for a reasonably useful mu package seemed to be the pythong qtchart stuff and pygame zero.

I started looking at the pygame zero stuff but ran into a bunch of files of questionable provenance among documentation and examples. I raised this with the author of pygame zero, first by private email and then was later asked to take it to a github issue which I did . Unfortunately things then went quiet. I have now decided to try and package pygame zero without the offending files.

skitt, 2018-06-07:
@russel I don’t have anything yet, I’m still working on the basic package construction (which, thanks to the intricacies of PyQt5 and the Debian Python policy, is “interesting”).

skitt, 2018-06-08:
I’ve pushed the packaging to https://salsa.debian.org/skitt/pyqt5chart; it works for me, there’s just an issue with one of the debug files I need to check.

russel, 2018-06-08:
@skitt I have forked and cloned. In my usual be as literal and irritating as possible mode:
```
|> python3 configure.py
/bin/sh: 1: ./sip: Permission denied
Error: './sip -V' did not generate any output.

|> which sip
/usr/bin/sip
```
Is this just a question of removing the `./` so as to get PATH lookup? I'll progress this and look to provide a pull request.

skitt, 2018-06-08:
You need to build it as a Debian package, ideally using `git-buildpackage`:

    gbp clone https://salsa.debian.org/skitt/pyqt5chart
    cd pyqt5chart
    gbp build-package -us -uc

This will clone the repository and set everything up so that the archives can also be found for the build. The last step will probably complain about missing dependencies; you’ll need to install those and re-run the `build-package` step.

russel, 2018-06-08:
I'll try that after lunch, gbp installed trivially. The implication here is that the README is not actually giving the right instructions for installation.

skitt, 2018-06-08:
Right, and that probably seems weird if you don’t work with Debian source packages all the time ;-). The idea is that Debian source packages contain all the upstream source, including `README` etc., and all the packaging files go in `debian`. We don’t adjust upstream instruction to match Debian build instructions (which are always the same in the end: retrieve the source package, and run `dpkg-buildpackage`; `gbp` is a convenient wrapper for git-hosted repositories).

russel, 2018-06-08:
I did say it was a long time since I had done this stuff. :-)
```
|> gbp clone https://salsa.debian.org/skitt/pyqt5chart
gbp:info: Cloning from 'https://salsa.debian.org/skitt/pyqt5chart'
|> cd pyqt5chart/
|> gbp build-package -uc -us
'build-package' is not a valid command.

Usage:
    gbp <command> [<args>]

The most commonly used commands are:

    buildpackage - build a Debian package
    import-orig  - import a new upstream tarball
    import-dsc   - import a single Debian source package
    import-dscs  - import multiple Debian source packages

Use '--list-cmds' to list all available commands.

|> gbp buildpackage -uc -us
gbp:info: Creating /home/users/russel/BuildArea/pyqt5chart_5.10.1+dfsg.orig.tar.gz
gbp:error: Error creating pyqt5chart_5.10.1+dfsg.orig.tar.gz: Pristine-tar couldn't checkout "pyqt5chart_5.10.1+dfsg.orig.tar.gz": execution failed: [Errno 2] No such file or directory: 'pristine-tar': 'pristine-tar'
```

skitt, 2018-06-08:
Oops, sorry about that, I’ve fixed it above. You also need to install `pristine-tar`...

russel, 2018-06-08:
OK, done, I am moving :-) There were various python3.*dbg packages I didn't have installed, I installed them and the process got going. However:
```
cd dbg-build-3.6 && python3.6-dbg ../configure.py --verbose -q /usr/bin/qmake -c -j 10 \
		-d /usr/lib/python3.6/dist-packages \
		--sip-incdir /usr/include/python3.6dm \
		--debug --no-sip-files
Error: Unable to import PyQt5.QtCore. Make sure PyQt5 is installed.
```
and yet:
```
|> python3
Python 3.6.5 (default, May 11 2018, 13:30:17) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyQt5.QtCore
>>> 

```

skitt, 2018-06-08:
Believe it or not, I did build the package in CI before pushing it, so it should be buildable ;-). Your build is failing because `python3.6-dbg`, the debug-symbol-equipped Python interpreter, can’t find `PyQt5.QtCore`, which is provided by `python3-pyqt5-dbg`. To verify locally, you need to run your `import` test in `python3.6-dbg`, not plain `python3`.

ntoll, 2018-06-08:
@russel @skitt you folks are awesome. Thank you for the effort you're putting into this. :-)

russel, 2018-06-08:
Ah of course python3.6-dbg is a completely different installation silo to python3.6, rather than just being the debug symbols for it. It seems though that python3-pyqt5-dbg package installs into the python3 silo not the python3.6-dbg one.

skitt, 2018-06-08:
They both install in `python3`... Doesn’t it work for you after installing the `-dbg` package?

russel, 2018-06-09:
It seems not:
```
|> aptitude search ~i | grep pyqt5
i  pyqt5-dev - Development files for PyQt5
i A python3-dbus.mainloop.pyqt5 - D-Bus Qt main loop support for Python 3
i A python3-pyqt5 - Python 3 bindings for Qt5
i  python3-pyqt5-dbg - Python 3 bindings for Qt5 (debug extensions)
i  python3-pyqt5.qtquick - Python 3 bindings for QtQuick module
```
```
|> python3
Python 3.6.5+ (default, Jun  8 2018, 21:55:12) 
[GCC 8.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyQt5.QtCore
>>> 

|> python3.6-dbg
Python 3.6.5+ (default, Jun  8 2018, 21:55:12) 
[GCC 8.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyQt5.QtCore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /home/users/russel/.local/lib/python3.6/site-packages/PyQt5/QtCore.so: undefined symbol: PyModule_Create2
>>> 
```

skitt, 2018-06-09:
ImportError: /home/users/russel/.local/lib/python3.6/site-packages/PyQt5/QtCore.so: undefined symbol: PyModule_Create2

means there’s a mixture of local (`pip`-managed?) modules and system modules; for this build Python should *only* be using system modules.

russel, 2018-06-09:
OK, that is because I installed Mu via "pip install --user". Whilst I never pip into the package managed area I keep forgetting it precedes the system stuff. It would have helped if I had actually read that line wouldn't it. :-)

russel, 2018-06-09:
Right, having removed Mu and all the dependencies pip3 put in place (which are exactly the ones I have installed via packaging, gggrrr…) I now have a complete build. I guess I should install with dpkg… duly installed and removed,so it is looking fine.

Sorry to have caused issues with this. But all good to go!

[34 more replies]

closed 2020-09-29 by carlosperate