mu-editor/mu #61: Really weird Pi 2 keyboard bug

carlosperate, 2016-01-13:
I am not quite sure how to replicate this specific issues, which I woudn't be surprised if it's not related at all with Mu, but something else going wrong.

Basically I've just done a clean install of raspbian on a Pi2, and installed the required PyQt5 packages, a VNC server, and nothing else. So, when I run Mu the keyboard does not type what it is expected, almost like a different layout is loaded, even though it works perfectly fine on any other app I've tried on the system.

I've created this quick video to illustrate me trying to type `display.scroll('test')`:
https://www.youtube.com/watch?v=L8lnCOmu8EY

ntoll, 2016-01-13:
Do you think this is a Heisenbug..? I've got the editor working on my RPi B+ and it works fine.

carlosperate, 2016-01-13:
I think this in issue with the VNC software, experiencing the same problem in a UDOO Neo board (cortext A9, so ARMv7 as well) running an Ubuntu based distro, when going through the remote desktop, but not when connected directly.

carlosperate, 2016-01-13:
Yep, it works when connected through a monitor, but I still encounter the issue on both boards when using the VNC server. Tried using a different client and the same problem persists,  I also tried tightVNC server from windows but I couldn't replicate it there.

I had a quick google around and didn't really find anything related (difficult to define for a concise search query), I doubt there is much that can be done, so feel free to close it.

ntoll, 2016-01-13:
OK... classing this as a VNC problem.

closed 2016-01-13 by ntoll

carlosperate, 2016-01-16:
Just in case somebody looks at this in the future, it seems like QT requires the xkeyboard extension on xserver, which is not provided during remote desktop (tested with tightvncserver and xrdp). This could possibly be "fixed" on he user machines if the correct keyboard configuration is provided (I believe via environmental variables).

https://bugreports.qt.io/browse/QTBUG-44938