mu-editor/mu #62: Exit message with unsaved changes can be confusing

carlosperate, 2016-01-14:
The current warning message displays "You have un-saved work!" with the options "Cancel" and "ok". The first time I encountered it I had to think about it twice before realising that "ok" would just allow me to exit without saving the changes.

Instead of changing the button message, which would require modifications to the interface, I suggest to just change the message for now. How about something along the lines of "There is un-saved work, exiting the application will cause you to lose it" ?

ntoll, 2016-01-14:
Good call, and I agree. Will make the changes in a mo.

ntoll, 2016-01-14:
Fixed by 807456d702145b5776638d92d46f37368280c8d2

closed 2016-01-14 by ntoll