mu-editor/mu #68: Mu failing to detect Micro:bit on Linux Mint

amos1969, 2016-01-20:
Hi

I cloned the repository, followed the instructions and ran Mu on my Linux Mint laptop. I could access and edit the Python file I wanted. 

When I clicked on the Flash button I got an error mrssage saying that no Micro:bit was detected. 

I tried this with the Micro:bit connected prior to running Mu, connecting it after Mu was started, pressing the reset button on the Micro:bit and disconnecting and reconnecting it. I was successfully able to transfer hex files I had created using the online stuff and one that I created from a python file using the tools in the micropython folder.

At no point did I tell Mu where my Micropython folder was, nor is it in the path. Does Mu need a working Micropython hex file, to function?

Cheers
Dave Ames

ntoll, 2016-01-20:
Hi Dave,

That's weird. Do you have the "mount" command installed..? If you look here:

https://github.com/ntoll/uflash/blob/master/uflash.py#L150

You'll notice that to detect the micro:bit on unix-y systems uflash.py (which mu uses) checks the names of mounted partitions that contain the word "MICROBIT" (which is how it should appear on your filesystem). It works perfectly for me on Debian Jessie. Perhaps you have a strange filesystem mount arrangement? More details about how you're set up are required for me to be certain.

The bottom line is, try using uflash and experimenting with the `find_microbit` function (http://uflash.readthedocs.org/en/latest/#uflash.find_microbit) and let me know what you see. Feel free to paste things as comments to this ticket.

One other thought... could it be a permissions related problem..? (A long shot, but you never know and I'm just guessing.)

Happy to help in any way that I can.

N.

amos1969, 2016-01-20:
Aha. I've successfully got Mu to Flash the microbit.

On KDE in Mint, it gives you the option to mount a drive when it's plugged in (ie to open it in Dolphin) but doesn't auto-mount the drive until you do. If I open the microbit as a drive, then I can flash the file from Mu successfully. But then it shows up as a new drive after flashing, which is not mounted.

Unfortunately that means that every time I want to flash a python file from Mu to the microbit, I have to manually open the "folder" first, which is less than ideal. I don't know whether this is Mint specific, KDE specific or inherited from Ubuntu. It also means that I end up with multiple MICROBIT entries in Dolphin (see screenshot).

[image]

Cheers
Dave

ntoll, 2016-01-20:
This appears to be a problem with KDE/Mint and not anything to do with Mu. I'll leave this open so when I write up the documentation I don't forget to mention it.

amos1969, 2016-01-20:
I agree. It isn't a Mu problem necessarily, but does need flagging.

gquintana, 2018-05-14:
I am running Mu 1.0b15 on Mint/Cinnamon and I can flash. In short, it's not related to Mint, maybe KDE?

ntoll, 2018-05-14:
OK... so since this is beyond the control of Mu (it appears to be a KDE "feature") I guess the resolution is to document this somewhere with a work-around. The docs will be in the website (source code here: https://github.com/mu-editor/mu-editor.github.io), so I've created an issue in that repository (https://github.com/mu-editor/mu-editor.github.io/issues/23) and will close this one.

closed 2018-05-14 by ntoll

timrichardson, 2019-07-10:
This issue features in Google searches. KDE has a setting ('control panel') Removable Storage, which has an option to automatically mount removable media (including an option to only do this after a user has first manually mounted it). 
When this setting is enabled, the MicroBit mounts to /media/$USER/microbit without intervention.