mu-editor/mu #72: SaveAs

cymplecy, 2016-02-03:
Might be me but didn't see a way to save an existing file under a different filename (for version increments etc)

ntoll, 2016-02-03:
It's not you. There will be simple versioning built in to an upcoming release. We're purposely trying to limit what you can do to the most simple actions. Once you're starting to feel the pain of "but I need feature X" then you're probably advanced enough to want to use a "proper" editor. ;-)

Happy to take suggestions on solutions if you disagree. This editor will only succeed if it lets its users work.

cymplecy, 2016-02-03:
:)
I'll wait for the versioning release then.
I'm all for keeping things as simple (si) as possible :)
I'm LOVING it as an Microbit IDE with the REPL and flash options.
Its obvious that its great for writing small little scripts :)

NDevox, 2016-02-05:
There's been some discussion about this on #57 as well.

ntoll, 2018-03-23:
So, double-clicking a tab's name allows you to rename.

closed 2018-03-23 by ntoll