mu-editor/mu #74: Drop file on support

gbaman, 2016-02-06:
It would be great to be able to simply drop a .py file onto the application in the menu bar. Although not a majorly important feature, it would be great to see it there. Have had a few times so far where I have gone to drop a script on it and then remembered I can't just yet.

NDevox, 2016-02-09:
Really nice idea actually.

Looks like there is a dropped event in PyQt. So we could implement something like this: http://stackoverflow.com/questions/4151637/pyqt4-drag-and-drop-files-into-qlistwidget

Where we take the file path, check it exists and then open a new tab with the file on it.

askvictor, 2016-08-07:
For bonus points: implement drag-and-drop for the file pane (i.e. from the OS file manager to the project folder or the device)

ZanderBrown, 2018-03-13:
Played a bit and it seems quite doable, just got to get my head around how the relationship between `EditorPane` & `Editor`. EditorPane is the obvious place for receiving drop events but Editor is responsible for loading files, unfortunately, EditorPane doesn't seem to have a reference to it's Editor

More to play with!

ntoll, 2018-03-13:
OK... there's a way in Qt to register event handlers with events (signals) emitted by a widget. The `EditorPane` will emit a signal along with the appropriate context when a file is dropped onto it. However, from the `Editor` (which has a reference to the `EditorPane` object), you'll be able to register a function *in the* `Editor` to handle the event from the pane. In this way the UI logic (in the pane) is separated from the logic in the editor (handling the drop event).

Does this make sense?

ZanderBrown, 2018-03-13:
That makes sense

ZanderBrown, 2018-03-22:
One was a pain to write but I weirdly enjoyed writing the tests for this

ntoll, 2018-03-23:
WRT Travis... pyflakes is complaining that an import is unused... I'll fix this up as part of the review.

@ZanderBrown heh... writing tests can be plenty of fun. ;-)

ntoll, 2018-03-23:
Better late than never... many thanks to @ZanderBrown for the work! :-)

PR #406 (2018-03-23): Drop files
Fix #74 (only 2 years late...)

If a file is dropped on the editor (doesn't listen on panels) a signal is bubbled up and then handled by opening the file

However the signal is only fired if the dropped file is `*.py or *.hex` (case insensitive) for everything else the current behaviour is preserved

Also included is a styling change following on comments on 51847232b37f7266e2fbb4108da23d9ebf64c4b0
(Yes i know it shouldn't really be in this PR)
closed 2018-03-23 by ntoll