mu-editor/mu #75: REPL doesn't work with automatically built binary on Linux.

ntoll, 2016-02-06:
What it says. When run from source locally, it works fine.

I'm using vanilla Debian Jessie with the expected QT libraries installed.

ntoll, 2016-02-06:
@carlosperate any idea what might be going on..?

carlosperate, 2016-02-07:
Haven't tested the REPL under Linux directly, most tests I've done were
through a virtual machine. I'll check it out later tonight once I get a
chance.

carlosperate, 2016-02-07:
It looks like mu needs sudo access for the REPL, in my test system (virtual machine running the minimal version of Ubuntu 15.04 64bit with xfce) I needed it for both the pre-built version and running locally. Can you run locally without sudo?

ntoll, 2016-02-07:
Working from source works fine without sudo if you're in the right group for USB/serial communications. I tried running it *as* sudo and it still didn't work (I checked with running from source immediately after and it *did* work). This suggests something amiss with the binary.

carlosperate, 2016-02-07:
I'll see if I can install a Jessie on a vm and replicate tomorrow. Since the issue might be specific to the CI server build environment, I'll create a temp branch to create builds that print the specific exception on console.

kushaldas, 2016-02-08:
For me in Fedora it did not work even from the source.

kushaldas, 2016-02-08:
On my Fedora 23 box:

>>> ports = QSerialPortInfo.availablePorts()
>>> for p in ports:
...   print p.manufacturer(), p.serialNumber(), p.description()
... 
MBED 9900023431864e450003100e0000003b00000000cc4d28bd MBED CMSIS-DAP

kushaldas, 2016-02-08:
The product and vendor identifier is coming as 

39985 and 32902 in Fedora 23.

carlosperate, 2016-02-09:
I was finally able to replicate the issue on a debian virtual machine. Looks like the problem was due to the baud rate set before the port was opened, a bit more info on the PR submitted: https://github.com/ntoll/mu/pull/80

PR #80 (2016-02-10): Correct serial configuration order
Fixes #75.

Even though the baud rate should be able to be set before opening the port (http://doc.qt.io/qt-5/qserialport.html#baudRate-prop), a few Stack Overflow qserialport answers indicated that it should be set afterwards. Doing this simple change fixes the REPL in the Linux CI-built binaries.

This binary was created on a test branch when debugging this issue, which should now work: https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/linux/mu-2016-02-08_23_49_10
closed 2016-02-10 by ntoll