mu-editor/mu #76: Latest OSX build crash

andreagrandi, 2016-02-07:
Hi, I wanted to try this build on my Mac (running El Capitan) but it crashes when I try to open it https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/osx/mu-2016-02-06_19_28_14.zip

I had to restore a previous (January) build to keep using the app.

Is there a way to send you some logs to better understand the problem?

inglesp, 2016-02-07:
I can reproduce this on 10.9.5.

inglesp, 2016-02-07:
However, everything works as expected when I run a newly-built app.

andreagrandi, 2016-02-07:
@inglesp how can you build an app if the editor crashes?

ntoll, 2016-02-08:
Could it be this..? https://github.com/ntoll/mu/commit/1f0d87ced298e53eb128bf474890b30593d030b6

ntoll, 2016-02-08:
OK... please try the latest build from this morning (currently in progress).

inglesp, 2016-02-08:
I've just tried with mu-2016-02-08_08_39_12.zip, and the app still crashes when I open it.

inglesp, 2016-02-08:
@andreagrandi -- to build the app, you need to follow the instructions in https://github.com/ntoll/mu/blob/master/package/README.rst.  Give me a ping on IRC if you try this and have any problems.

inglesp, 2016-02-08:
The problem seems to have been introduced between 2016-01-22_17_16_00 (which works) and 2016-01-26_08_09_26 (which doesn't).

This crept in in build [114](https://travis-ci.org/ntoll/mu/builds/104837788), which as far as I can tell differs from the previous build [113](https://travis-ci.org/ntoll/mu/builds/104136624) by the following: https://github.com/ntoll/mu/compare/84e37b979bf5...5486c16fe2c6.

This just looks like docs changes to me!

andreagrandi, 2016-02-08:
@inglesp I will be able to triage this when I go back home this evening

inglesp, 2016-02-08:
Could this change to docs/conf.py have anything to do with anything?  https://github.com/ntoll/mu/compare/84e37b979bf5...5486c16fe2c6#diff-85987f48f1258d9ee486e3191495582d

inglesp, 2016-02-08:
Assuming it's not the code that's changed, but something to do with the environment, I've noticed the following differences between the output of build 113 and build 114:

 * 113 uses a version of Python 3.5.1 with 3,420 files, while 114's version has 3,438 files.  I suspect these extra files relate to a new version of pip.
 * 113 uses pip 7.1.2, while 114 uses pip 8.0.2.
 * 113 uses sqlite 3.10.1, while 114 uses 3.10.2
 * 113 uses pytest-2.8.5, while 114 uses pytest-2.8.7
 * 113 uses sphinx-1.3.4, while 114 uses sphinx-1.3.5
 * In 113 the new dist directory contains 36,004kb, while in 114 it contains 35,972kb.

These all look harmless, except for the last line, which may indicate that the problem relates to some missing files.

inglesp, 2016-02-08:
https://github.com/ntoll/mu/commit/31898b027e3406da9a5f293f93b4e065cb63a2ff is a Hail Mary commit to see whether the path munging in the docs config does indeed affect the build process somehow.

inglesp, 2016-02-08:
This built mu-2016-02-08_11_40_29.zip, which still failed to run.

inglesp, 2016-02-08:
Ah, progress!

If I run the `mu` executable that's inside the package, I get the following:

```
~/Downloads/mu-2016-02-08_11_40_29/mu.app/Contents/MacOS $ ./mu 
Traceback (most recent call last):
  File "<string>", line 11, in <module>
  File "/usr/local/lib/python3.5/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
    exec(bytecode, module.__dict__)
  File "pkg_resources/__init__.py", line 48, in <module>
  File "pkg_resources/extern/__init__.py", line 60, in load_module
ImportError: The 'six' package is required; normally this is bundled with this package so if you get this warning, consult the packager of your distribution.
pyi_rth_pkgres returned -1
```

inglesp, 2016-02-08:
https://github.com/pyinstaller/pyinstaller/issues/1789 might be relevant.

inglesp, 2016-02-08:
Since this bug mentions pytest, and since we've noticed that the version of pytest changed when the app broke, I'm going to pin the dependencies in requirements.txt and see if that helps.

inglesp, 2016-02-08:
Pinning dependencies hasn't worked, so I'll revert that.  (It's something we should investigate doing though.)

Reading the linked bug report more carefully suggests we should pin the version of setuptools we use, so I'll add that to requirements.txt.

inglesp, 2016-02-08:
Yes, that's fixed it!  mu-2016-02-08_12_37_54 is a working package.

ntoll, 2016-02-08:
:clap: :+1: :-)

inglesp, 2016-02-08:
Have opened #77 and #78 to track work to make debugging this kind of thing easier in future.

inglesp, 2016-02-08:
Issue accidentally re-opened while cleaning kiwi fruit juice off my keyboard!!

closed 2016-02-08 by inglesp

ntoll, 2016-02-08:
You realise that in 2000 years time some sort of digital archaeologist is going to find this ticket and try to imagine you cleaning kiwi fruit juice off your rudimentary input device, the so-called "KEY-BOARD". They'll boggle at how we ever managed to use QWERTY. ;-)