mu-editor/mu #81: User Help

alistair-broomhead, 2016-02-09:
Perhaps worth thinking about for post V1.0, is some sort of onboarding tutorial, whether it's 

1. a guided tour through the application, 
2. a "tip of the day" feature, or 
3. just a set of help pages that guides you through everything from a beginner's eye view. 

These aren't mutually exclusive. `3.` is possibly the easiest, `1.` is possibly the best for discovery, and `2.` isn't a bad half-way house ;)

NDevox, 2016-02-09:
I think all 3 of these would be great.

alistair-broomhead, 2016-02-25:
It may be worth bearing in mind [techniques used in games](http://lunar.lostgarden.com/Mixing_Games_and_Applications.pdf), as mentioned in #90

alistair-broomhead, 2016-02-25:
@ntoll Having looked a bit further through the issues, is this a duplicate of #32? Should we close this and copy in any extra ideas, or take the ideas expressed in '32 out of this issue?

ntoll, 2016-03-16:
Simplest possible solution here:

https://github.com/ntoll/mu/pull/98

Love the idea of an on-boarding tutorial (love the link to the techniques used in games).

;-)

alistair-broomhead, 2016-03-16:
@ntoll I just struggle sometimes with how games have had these techniques for keeping users engaged for decades, while other software ignores even the basic learnings in terms of difficulty curves and pacing. 

Hoping to have a bit more time to contribute to this by late spring/summer and would love to so something along these lines one idea I've been playing with is having fairly extensive configuration (which could be imported/exported) and having the tutorial show you a feature that's not enabled and give you the option of whether to turn it on or not, which would allow you to go from a clean and simple UI with just the very basics, to something that, while still as clean and simple as it can be, has a few added features for power users. 

The important thing would be that beyond the first-boot features, there was never an enabled button or keyboard combination that hadn't been explained to you. There could then be links in the help which allow you to launch the tutorial for a given feature, as well as a guided exploration mode, such as a button in the tool-bar labelled «show me something new». 

The tip-of-the-day could then show you info about features you have already enabled, or randomly show you a new feature. If we were to instrument this we could collect metrics when people turn on a feature, to get an idea of the order in which people add features and the average timeline, and figure out if there was anything that was unbalanced (usage differs from our expectations relative to other features) in order to improve the on-boarding process.

This is all reasonably ambitious, none of this would be particularly easy and it wouldn't all have to be done to start getting value, but it seems like it could be the most sure-fire way to achieve the project's goals of being a brilliant editor for teachers and students; it would also allow is to add requested features without compromising the core experience.

NDevox, 2016-03-16:
@alistair-broomhead I'm in complete agreement.

I watched this a few months ago (https://www.youtube.com/watch?v=MMggqenxuZc) and had a similar thought, why can't learning to use an app be more natural and fun. (admittedly the video is more extreme in comparison).

If we can steadily show features instead of overwhelming they might take them on more readily.

As an aside, @ntoll  have we released at v1 yet? I have specifically stopped working on this and focused more on other projects as I believe we were going to push for V1 and do a feature freeze.

Avaren, 2016-03-17:
@ntoll @alistair-broomhead @NDevox without seeing this I just posted something over at #32. Thoughts?

[image]

alistair-broomhead, 2016-03-18:
It's not what I'd want for the final UI, but it would be a good start and let us test the content. if each quest had a screenshot it would be the sort of thing we'd want in a tip-of-the-day

Avaren, 2016-03-21:
Added #100 that has my initial ideas

ntoll, 2016-03-21:
Cool... let me get through this week (with the launch of the device tomorrow) and then we can move forward on all this stuff! :-)

Very exciting.

johnkershaw, 2016-06-18:
Can this happen soon? I'm a Computing teacher (15 years) and a programmer (30 years) who's using IDLE, Khan, codingbat, codecombat, sololearn, codewars and others with students learning Python and Javascript. Feeling the pain of unfriendly environments (Khan JS/Processing is still the kids' favourite) I downloaded mu with great anticipation. 

And got stuck. Immediately. 

I'm no idiot, but this was my 'learning journey':

Web site: absolutely no information. Links to download for 3 platforms, nothing else. Okay...

App: a single Mac OS executable (no help/config/anything files). Downloaded, ran it.

Interface: 9 buttons, including zoom in/out, but no Help button. No menus. Right-click? Nothing.

Gave up and came here. For an editor aiming to be super-accessible, I'm finding the EXACT opposite. Did I come in through the servant's door or something? 

Super weird.

ntoll, 2016-06-18:
Hi @johnkershaw,

Thanks for the feedback. I wrote Mu.

Please remember that Mu is completely written in my free time. If things are missing, it's simply because I've not yet found the time to make things happen. Since this is an open-source project, please feel free to jump in and send me PRs.

You're right! It's supposed to be simple and accessible. The important thing you forgot to add to that sentence is "for beginners". You mention you have 30 years experience as a programmer, so I imagine you're used to an IDE, vim or emacs. Mu is the antithesis of this - feedback from many teachers as well as friends at the Raspberry Pi Foundation has told us that a restricted set of functionality, accessible via obvious buttons is the ideal environment for *new* programmers. No confusing menus, no hidden right clicks, no need to configure it. Put simply, what you see is what you get, and it "should just work".

Ergo, if you're thinking, "but why don't they have feature X?" then you're probably too advanced for Mu! ;-)

However, you're right about missing help. It's coming and I hope to find time over the coming weeks to make progress. As I mentioned, this is something I do in my spare time and I'd certainly appreciate help (your feedback is super helpful by the way).

Hope this explains things.

N.

johnkershaw, 2016-06-18:
All I'm really looking for is code completion, as a way of transferring students from block-world to text-world. Blocks & categories do a great job of exposing all the 'atoms' available, whilst examples show what 'organisms' others have created using those 'atoms'. Micropython has lots of shortcuts but I'm not sure how helpful they are - typing 'cl' instead of 'class'? How much time will that save me? I'm thinking of the way text adventures allow you to start from almost nothing (look = current location, examine items mentioned in the description to make the player aware of new objects they can further manipulate, right-clicking an object in-game might give you a list of verbs that can be used with it....)

Disclaimer: I've been teaching Computing in Cambodia (and remotely in the UK) for the last two years, but am over in the UK for the summer (5 weeks) so only touched a micro:bit for the first time 3 days ago! I'm fast-tracking whilst I'm here, hoping to take a class set back to Cambodia if they become available soon enough.

And I totally appreciate that you're making this in your spare time, top bloke, good egg, etc. Hope the 'new user trying to figure stuff out from blank canvas' is helpful. I may push something, but pretty new with 'git' too.

John.

ZanderBrown, 2016-06-18:
definitely a good point with auto compleate. being able to type display. and get a list of 'get_pixel, set_pixel...' possibly with a short description alongside would help people discover the various functions available without having to reference to un-child-friendly docs like https://microbit-micropython.readthedocs.io/en/latest/display.html

johnkershaw, 2016-06-18:
But a kid wouldn't even know they had to type display. - I didn't!

You already have a template for new documents - why not expand that template to demonstrate a useful basic template that contains lots of examples of scrolling text, displaying images, getting button & pin inputs, etc, with simple comment notes above each, like a standard Unix config file.

ZanderBrown, 2016-06-18:
from commits I've seen this is being worked on. just to point out here I don't do any coding work on Mu and those that do only work on it in their spare time and from my work attempting to clone it in Gtk3 this is not an overly simple project from a coding perspective. I think to a certain extent the theory is that as Micro:Bit is primarily an in school (at least at first) project students first experience of it will be in a classroom environment where a, hopefully, experienced teacher such as yourself will be present to teach the language while Mu essentially keeps out the way letting them play

ntoll, 2016-06-20:
So, I've got a branch in progress that gives you:

* Online help to be hosted at http://codewith.mu/ and accessed via an obvious "Help" button.
* Auto-completion. If you type `display.` you get to see `scroll`, `show`, etc...
* Call tips. As you start to type a function - e.g. `display.scroll(` - you get to see the arguments the function takes and a short, child friendly description of what it does (based upon the built-in help in the REPL).

...as well as various other improvements, bug-fixes and simple new features.

As for not knowing what to type, I'm afraid I simply expect a teacher to know what they're doing. :-) Of course, people are new to MicroPython so I've written a _very simple_ yet extensive series of tutorials for teachers: https://microbit-micropython.readthedocs.io/en/latest/tutorials/introduction.html (these tutorials will be linked to from the on-line help).

As always, feedback most welcome. :-)

johnkershaw, 2016-06-20:
This sounds spot on, you've nailed all the possible locations where a kid can get stuck. IDLE has code-completion, tool-tips etc, but without the examples, how do I even know what functionality the module has? Excellent.

A teacher WILL know what to type, but the whole point of the micro:bit is that kids have been given them, to take home and play with. There's no teacher there; they're on their own. Whatever they're using needs to be learnable by (determined) exploration, with no expectation of outside assistance. 

Also, the situation in the UK at the moment is that the vast majority of IT teachers are NOT programmers - 12 months ago the most they knew was fancy Excel functions. They've been learning as fast as they can, but in many cases they are out-stripped by their students, who have time, youth and curiosity on their side.

Harking back to the days of the BBC Micro, the 'online help' was the big fat 'User Guide' which had a comprehensive explanation of every available BASIC command, with contextualised examples, and even covered lots of stuff we might now consider 'too hard' such as intro to assembler and the *FX commands. But kids learnt very quickly and reached those 'Advanced' stages sooner than anyone expected.

Above the whiteboard in my classroom it says, "Don't let your teachers limit your learning." It's a constant reminder to them that what we do in class is a taster, it should be just the beginning of their exploration. Some of the stuff my students have created with very limited knowledge has literally dumb-founded me. Without knowledge of loops or arrays, they make amazing things from pure imagination, unstifled by all the things we 'know'. It's long-winded and messy, like our early forays into goto (!) but it's THEIRS. That's important. Correct, succinct, industry standard can come later, after they've explored and got themselves an appetite, a taste for it.

I worry that in some ways the level-by-level training in modern games gives too much structure, too little freedom. Minecraft, open-world and physics-based games allow kids to ask themselves "What do *I* want from this environment?", rather than merely following the developer's breadcrumb trail. That's a key aspect to learning.

carlosperate, 2016-06-20:
So is the documentation going to be kept online for the moment? I assume each doc output version will be kept in the website on a separate folder, and the help link generated from the the version indicated by the `__version__` property? 

Will the general micropython documentation be included? Or will a link be added to the Mu documentation?
In the case of including a link we run the risk old local user versions of Mu linking to an erroneous version of the documentation until the microbit micropython starts tagging their docs.
On this note, perhaps the micropython version embedded intro uFlash should be included in uFlash itself, and then we could have the documentation grab that to autogenerate a link to the correct micropython documentation?

ntoll, 2016-06-20:
@carlosperate you've pretty much nailed what I was thinking of doing. I believe RTD allows you to build versions of the documentation to which we can point for specific released of MicroPython. I'll investigate further and work out the simplest solution / path of least resistance.

ZanderBrown, 2016-06-20:
"Don't let your teachers limit your learning." I like that.

ntoll, 2016-06-20:
Heh... I believe it was Noam Chomsky he said that, "school is imposed ignorance".

Anyway... folks, I used to be a senior teacher, I'm well aware of the expectations of and challenges to teachers. I'm telling you this not because I want you to tell me I'm a "good egg" ~ although that's always nice ;-). Rather, to manage your expectations of what I'm up to. ;-)

Your feedback is invaluable and today I've managed to get most of the autocomplete and call tip work done. It'll definitely be in the next release. I believe I've got all the MicroPython-on-the-micro:bit API covered in this respect too.

ntoll, 2018-06-04:
The new website (coming soon) contains extensive tutorials and how-to guides, thus meeting the original point of this issue. Closing.

closed 2018-06-04 by ntoll