mu-editor/mu #82: Commands blanked from cursor in REPL

jrmhaig, 2016-02-10:
When executing a command in the REPL, if enter is pressed whilst the cursor is not at the end of the line then the part of the line after the cursor is blanked. The command does, however, execute correctly.

matt-land, 2016-06-03:
I investigated this, and added a trap on the new line characters in REPLPane.keyPressEvent:
if key == Qt.Key_Enter or key == Qt.Key_Return:
            return
With the trap, I successfully prevented the text stopped being garbled. I think this 'garbling' is from the round trip to the microbit, and not from mu.

I captured the serial communication to the microbit from the repl panel with this string: print("xx") [ and then backed the cursor up between the word 'print' and opening the parenthesis. ]
It looked like this:
[image]

The text being sent looks fine to me. The "\r" was the Enter/Return being sent. 

Since mu is opening a serial link, and that usually means displayed text is being echo'ed back, it seems pretty likely this is the microbit implementation, and not mu.

eduvik, 2016-08-12:
The \x1b[D are VT100 control codes; I've implement some of these at: https://github.com/eduvik/mu/blob/feature/multi-board/mu/interface.py

eduvik, 2016-10-09:
Should be fixed in https://github.com/mu-editor/mu/pull/167

closed 2018-06-04 by ntoll