mu-editor/mu #86: Default folder on Windows not in Documents

mavhc, 2016-02-12:
On my Windows 7 Pro 64bit machine (joined to domain, but local Documents) the python folder is created in c:\users\username\ not c:\users\username\Documents

jonwitts, 2016-06-07:
This is a real issue for our students as we have Windows "My Documents" being redirected to the network file server, but the path which Mu defaults to save in every time is on the local machine. It would be great if we could control this behaviour, or at least have the Windows version of Mu to default to a folder inside the Documents folder.

NDevox, 2016-06-07:
@jonwitts Good to know that this is an important issue.

I don't have much time at the moment but I will be teaching a class with mu the weekend after next so will try and fix the issue by then. No guarantee it will make it into a release by then though!

jonwitts, 2016-08-18:
It would be great if this could be fixed! Let me know when it makes it in to a release...

carlosperate, 2016-08-19:
Based on the comments by @jonwitts ([link](https://github.com/ntoll/mu/issues/87#issuecomment-240887660)) and @ZanderBrown ([link](https://github.com/ntoll/mu/issues/87#issuecomment-240946388)) in https://github.com/ntoll/mu/issues/87, it sounds like this should definitely be considered for inclusion as the default location in Windows. Having a quick look at my Windows dev machine, I can see that the "Documents" folder (at least in Win7 and Windows10) is used as the workspace area for quite a few IDEs as well.

ZanderBrown, 2016-12-25:
With #178 & #179 could this be considered resolved?

carlosperate, 2017-01-07:
I would probably say that the other issues were covering "being able to change" the default directory, while this one was more specific to change the Windows location to a better place.

Yes you could add a settings.json file next to the Mu.exe file and that should be able to change it to wherever you'd like, but it sounds to me like moving it from the user directory to the user documents directory is not only better for school users, but a common pattern for IDEs to have their workspace folder.

ZanderBrown, 2017-01-07:
Yes. I think it's in the right place for *nix but windows is being the awkward exception where the default should be inside Documents.

How about replacing the last line of `get_worspace_dir` from `return os.path.join(HOME_DIRECTORY, WORKSPACE_NAME)` to something like:

```python
if os.name == 'nt':
    return os.path.join(HOME_DIRECTORY, 'Documents', WORKSPACE_NAME)
return os.path.join(HOME_DIRECTORY, WORKSPACE_NAME)
```

Of course that would fail on Windows < Vista but as XP is end of life would this matter?

Due to the current implementation of settings this would make this the default for new users but old users would continue with the current behaviour (unless they changed / removed `settings.json`)

carlosperate, 2017-01-07:
The change should probably go into mu/contrib/appdirs.py instead.

ZanderBrown, 2017-01-07:
Why?

ntoll, 2018-06-04:
We use the `appdirs` module to discover what the "blessed" home directory is supposed to be on each machine/OS. If this is a problem with how appdirs works, please report to the upstream project. Closing.

closed 2018-06-04 by ntoll