mu-editor/mu #87: Previous load/save folder not being remembered on Windows

mavhc, 2016-02-12:
1. Press Load, c:\users\username\python folder opens
2. Move to another folder, load a .py file
3. Press Load again, back to c:\users\username\python, should have opened last opened folder

Same happens with Save

NDevox, 2016-02-12:
I have a feeling that this hasn't been worked in yet?

Here is an SO problem which suggests and answer:

http://stackoverflow.com/questions/29914909/pyqt4-native-file-dialog-remembering-last-directory

jonwitts, 2016-06-07:
Is issue  #86 related to this?

NDevox, 2016-06-07:
@jonwitts.

I doubt they are related strongly - only by the fact that they both involve windows save folders.

aeroaks, 2016-08-17:
Quick look into the code reveals that User's HOME_DIRECTORY is used to create PYTHON_DIRECTORY (user/python) which is referred to every call of load function.

If we want to keep the default to PYTHON_DIRECTORY then we can make the session directory variable that stores the path used by the user, or as @jonwitts pointed out, couple #86 and #87 (this issue) for a improved solution.

ZanderBrown, 2016-08-17:
A session var only survives for the current session so I'm not sure how that would help

aeroaks, 2016-08-18:
@ZanderBrown 
Just listing the possible scenarios. The one we decide on, I would be interested in implementing that!

carlosperate, 2016-08-18:
As already pointed out, the load and save default directory is manually set to PYTHON_DIRECTORY. Based on teacher feedback (https://github.com/ntoll/mu/issues/126#issuecomment-236389947) it's probably done like that by design, and I would argue that it's probably better to keep it that way. It might be a small annoyance to have that additional step for users that want to save their code somewhere else, but it helps ensure code is generally saved on a "known location" for the users that might just save it wherever that prompt windows happens to be displaying.

mavhc, 2016-08-18:
Windows apps seem to work like this:
Load app, default folder for opening is Documents (I'm fine with that being Documents\python)
Open a file from another folder.
Open file picker again, it's in the new folder again.
Close the app
Open the app, defaults back to Documents

carlosperate, 2016-08-18:
You are completely right, that is how most applications behave, and what most users would expect.

The point I was trying to make is that the current implementation, while it might be a bit annoying, is probably trying to facilitate and hint towards the use of a single, known directory, for most of Mu work. I personally see this as an acceptable trade off, as it is not really limiting the areas where you can save and open from, but it persistently suggests the same directory.

jonwitts, 2016-08-18:
The issue with this on a Windows network (as most schools using this software will be) is that the default home directory is local to the PC the student is using; however HOME_DIRECTORY\Documents is *normally* redirected to a network file server...

By having Mu save into the Documents sub-directory most Windows users will find their files for Mu saved on the network not the local PC.

In a school environment this is a real necessity as students will not use the same computer evry lesson and invariably forget to change the default save path.

carlosperate, 2016-08-18:
Yes, and based on the conclusion from https://github.com/ntoll/mu/issues/126 the option to select a custom default location should be added.

Making the PYTHON_DIRECTORY default inside of the Documents folder I guess still needs to be discussed. To be honest I am a bit surprised only **that** folder gets saved in the network instead of the entire user user directory, @jonwitts do you know if that is a common pattern, rather than being a specific configuration from your network?
If it's the second we should probably leave the default as it is and relay in the configuration file to customise it.

jonwitts, 2016-08-18:
@carlosperate Using the Microsoft Group Policy tools for folder redirection (the standard method to redirect user folders from local to network drives) the only option is to redirect the "My Documents" folder, not the entire user profile folder.

It is possible to redirect the entire folder but that would require custom scripting by the MS sysadmin rather than using the built-in MS tools.

I would say that it is a common / default pattern as opposed to one specific to my network.

ZanderBrown, 2016-08-19:
ignoring the fact that `python` isn' t a good choice the Windows save location should be in Documents not home because the average Windows user doesn't even know they have a home folder and as mentioned before most schools have a network storage setup resulting in Documents being redirected however the rest of home is not meaning students will loose work

carlosperate, 2016-08-19:
I hope you guy's don't mind to move this specific conversation to https://github.com/ntoll/mu/issues/86 (I have already [replied there](https://github.com/ntoll/mu/issues/86#issuecomment-241024995) as well). That way we can keep these issues organised.

carlosperate, 2016-08-19:
@ntoll could you clarify if this specific feature (save and load location always being PYTHON_DIRECTORY by default, not remembering the last used location) is still to be implemented, or if the current behaviour has been done by design?

ntoll, 2018-06-04:
I'm closing this since the default location can be overridden by the user. This is going to be documented in the new website (coming soon).

closed 2018-06-04 by ntoll