mu-editor/mu #88: Mu issues on Raspbian Wheezy

gbaman, 2016-02-15:
When trying to use Mu on Raspbian Wheezy, I get a ```GLIBC_2.15 not found``` error.
Although Raspbian Wheezy is no longer officially supported by the Raspberry Pi Foundation, many schools/organisation still use it due to lack of a direct upgrade path to Raspbian Jessie and will likely continue to do so till (at least for schools) the end of the school year.     
[PiNet](http://pinet.org.uk/) (used by hundreds of schools) is still Raspbian Wheezy only, although Jessie support is due within the next few weeks. As mentioned above though, many schools will want to stick with what they know works till the end of the school year anyway.   
Perhaps is worth targeting the build against an earlier version of GLIBC for the Raspberry Pi build?
Or include instructions for additional packages needed to be installed on Raspbian Wheezy.    
[image]

carlosperate, 2016-02-15:
The easiest way to do this would be to build Mu on Wheezy then. Is there any image download link available? I had a quick look online, but couldn't really find an `.img` file.

ntoll, 2016-02-15:
I guess @ben_nutall would know where to get the right image. Alternatively, could you not just use a PiNet image?

carlosperate, 2016-02-15:
You probably meant @bennuttall.
Ideally I would like to use a vanilla Wheezy with no updates at all, just in case. Has the raspberry pi foundation released updated versions of Wheezy? If so, how could I identify the earliest image?

bennuttall, 2016-02-15:
As Andrew said, we've moved on from Wheezy and no longer support it. However, all previous images are available for download at http://downloads.raspberrypi.org/raspbian/images/raspbian-2015-05-07/

carlosperate, 2016-02-15:
Thanks @bennuttall!

In this case I'll be looking at the first release, so I'll try to build it with [2012-07-15-wheezy-raspbian](http://downloads.raspberrypi.org/raspbian/images/2012-07-15-wheezy-raspbian/).

gbaman, 2016-02-15:
Well, I don't think we need to go back that far...
I think building for the [final Raspbian Wheezy version from May 2015](http://downloads.raspberrypi.org/raspbian/images/raspbian-2015-05-07/) given that was out last school year and there is an upgrade path (using apt-get upgrade) to get to that point from most other images.
If schools were flashing new cards for their students over the summer, this is the image they would be using. It is just that there is no direct official upgrade path from Wheezy to Jessie.

ntoll, 2016-02-15:
Many apologise to @bennuttall - I went all "snake case" for no apparent reason. ;-)

bennuttall, 2016-02-15:
Is that your passive-agressive way of telling me my username isn't Pythonic? :P

ntoll, 2016-02-15:
No, it's me admitting I'm stupid. ;-)

carlosperate, 2016-02-23:
Sorry guys it's taking so long, had SD card corruption issues and then my old pi 1B stopped booting altogether (turned out to be the SD card connector becoming loose).

I've been trying to use a couple of different versions of wheezy, but I'm unable to install python3-pyqt5, can't find the package. Later tonight I'll try doing a general search in apt to see if the package is called anything different, or try to see if the pip package works, but if it doesn't I might have to build from source and that will probably take a little while longer to get together.

carlosperate, 2016-02-24:
Looks like I might have to build from source  :disappointed: :

```
pi@raspberrypi ~ $ apt-cache search pyqt
libpythonqt2-dev - Dynamic Python binding for the Qt framework - development
libpythonqt2.0 - Dynamic Python binding for the Qt framework - runtime
pyqt-tools - pyuic and pylupdate for Qt3
pyqt4-dev-tools - Development tools for PyQt4
python-guiqwt - efficient 2D data-plotting library
python-pivy - Coin binding for Python
python-qscintilla2 - Python bindings for QScintilla 2
python-qt3 - Qt3 bindings for Python
python-qt3-dbg - Qt3 bindings for Python (debug extension)
python-qt3-gl - Qt3 OpenGL bindings for Python
python-qt3-gl-dbg - Qt3 OpenGL bindings for Python (debug extension)
python-qt4 - Python bindings for Qt4
python-qt4-dbg - Python bindings for Qt4 (debug extensions)
python-qt4-dbus - D-Bus Support for PyQt4
python-qt4-dbus-dbg - D-Bus Support for PyQt4 (debug extensions)
python-qt4-dev - Development files for PyQt4
python-qt4-doc - Documentation and examples for PyQt4
python-qt4-gl - Python bindings for Qt4's OpenGL module
python-qt4-gl-dbg - Python bindings for Qt4's OpenGL module (debug extension)
python-qt4-phonon - Python bindings for Phonon
python-qt4-phonon-dbg - Python bindings for Phonon (debug extensions)
python-qt4-sql - Python bindings for PyQt4's SQL module
python-qt4-sql-dbg - Python bindings for PyQt4's SQL module (debug extension)
python-traitsbackendqt - PyQt backend for Traits and TraitsGUI (Pyface)
python3-dbus.mainloop.qt - D-Bus Support for PyQt4 with Python 3
python3-dbus.mainloop.qt-dbg - D-Bus Support for PyQt4 (debug extensions for Python 3)
python3-pyqt4 - Python3 bindings for Qt4
python3-pyqt4-dbg - Python3 bindings for Qt4 (debug extensions)
python3-pyqt4.phonon - Python3 bindings for Phonon
python3-pyqt4.phonon-dbg - Python3 bindings for Phonon (debug extensions)
python3-pyqt4.qsci - Python 3 bindings for QScintilla 2
python3-pyqt4.qtopengl - Python 3 bindings for Qt4's OpenGL module
python3-pyqt4.qtopengl-dbg - Python 3 bindings for Qt4's OpenGL module (debug extension)
python3-pyqt4.qtsql - Python3 bindings for PyQt4's SQL module
python3-pyqt4.qtsql-dbg - Python3 bindings for PyQt4's SQL module (debug extension)
```

```
pi@raspberrypi ~ $ lsb_release -da
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 7.8 (wheezy)
Release:        7.8
Codename:       wheezy
```

carlosperate, 2016-02-28:
So this turned out to be a much larger effort than I expected, and unfortunately after all this time I haven't really got that far.

Tried building Qt5 from source (quick tip, use at least a 16GB SD card or you'll get odd errors half way through), and ended up hitting some weird segmentation fault errors after about a little bit over a full day of compiling. I looked around and a few issues from the qt bug tracker indicated that it is probably a GCC bug and that I should update it (however the issues themselves were closed without any feedback from the originator to indicate if that fixed it... :confused:). The problem I then had was that the version required was not in the wheezy repository, and using jessie or the testing repo to update gcc version ends up updating the GLIBC as well, so can't proceed that way. At this point it looks like if I want to continue this way I might have to compile gcc, so that I can then recompile Qt5, to then build PyQt5 for the Python bindings to create the PyInstaller executable. It seems like a can of worms I'm not sure I want to open.

So then I though, since the initial issue seems to be with python3.4m.so, as long as there is no other issues with the Jessie PyQt/Qt glibc version being >2.13, then maybe getting just Python3 sorted could do the trick. So I got a hello world build in Jessie, moved it to Wheezy and I get the same error as expected. Turns out the latest python3 in wheezy is 3.2 and PyInstaller needs at least 3.3, so then I compiled Python3.4.2 (with `--enable-shared`) and the PyInstaller bootloader in Wheezy with the old glibc, moved it to Jessie, build it using PyInstaller, moved it back to Wheezy and it worked, hooray! Seemed like I was getting somewhere.  The problem here is that even if I remove all the python3 apt packages from Jessie, install my own version and replace all symbols/aliases etc for the system to use my version of python3; if I try to install PyQt5 using apt it installs the repository version of  python3.4, pretty much ignoring any other version on the system. I googled around to see if there is a workaround for that, and all solutions point to building from source.

As I last resource I tried doing a PyInstaller build from the "system default python3" and then manually replace the python libraries imported into the build with my "glibc2.13 python3" versions, but it ended up having issues finding lib files in the wrong directories and it looked like a very fragile fix to the issue anyway.

So that's my "progress" so far, it's taken a long time mostly because compiling on the pi takes a long time (as hard as it is, I didn't want to add cross-compilation to the pot). I'm kind of running out of free time to dedicate to this, but I am more than welcome to any suggestions. I might have a quick look into the feasibility of updating wheezy's glibc instead of the other way around, that might be simpler.

ntoll, 2016-02-28:
@carlosperate blimey matey... that was some rabbit hole you've fallen into. Given that people will be moving away from Wheezy and onto Jessie and think it might be opportune to simply say we don't support Wheezy (despite your best efforts). The bottom line is it's far easier for a user to upgrade to the latest and greatest (as they should be doing anyway), than it is for us to back-port to the older version of Debian.

Kudos x 10000000000 for all that effort!

Thoughts..?

carlosperate, 2018-03-17:
Taking in consideration it's been 6 years since Wheezy was released, and that PiNet is already available with Jessie I think we can leave this one as a "won't fix".

closed 2018-03-17 by carlosperate