mu-editor/mu #90: Find and replace

gbaman, 2016-02-16:
It would be great to see a "find" and "find and replace" option in Mu when hitting ctrl + f (cmd + f on Mac) similar to many other text editors, including IDLE.
I just went to replace a mistake in code I had made in a number of places, only to discover couldn't easily do it. Ended up copying over into another text editor to do it.

stestagg, 2016-02-17:
It's definitely not finished, or well tested yet, but I've had a go at starting a find-replace dialog on my repo: stestagg/mu@1f95b5425f14e8170308200f5bd20676d3d3a176

[image]

NDevox, 2016-02-23:
Would it be alright to hide the replace dialog? So that it is optional?

It is pretty common to want to find something without replacing and they are normally presented as separate things.

stestagg, 2016-02-23:
Yeah, I certainly thought about that, I'm not sure how to make it look to keep it simple, but retain the ability to do a replace tho.  Suggestions?

NDevox, 2016-02-23:
Could do tabs, like notepad++:
http://sqlblog.com/blogs/jamie_thomson/image_1AFC2B61.png

It's pretty common to see that.

I've also seen a hidden part of the dialog, which has a section like the following at the bottom:

`Replace----------------------------------------------->`

Where if you click on the arrow head the replace section comes into view.

But I think tabs are the clearest.

ntoll, 2016-02-24:
Hi... apologies for the silence at my end - I've been travelling.

This is great. But my *biggest* concern with Mu is complexification via featuritis. :-)

I think we most definitely *should* have find/replace. However, I don't think we should add a new button. If anything, it could be something accessed solely via hot-key.

The feedback from teachers and kids is clear (as reported by Carrie Ann at RPi Foundation). They only want the most basic and simple functionality presented in a clear and obvious way.

Search and replace, while being helpful, isn't really in this category of "most basic". So when it comes to such "advanced" features I think we should implement them via hot-keys. Adding more buttons breaks the "simple and obvious" aspect of Mu.

The bottom line is, if you're asking for feature X, then you're probably advanced enough to not have to use Mu. ;-)

Does this make sense? Would love comments, constructive critique and ideas..! 

:-)

stestagg, 2016-02-24:
Hi, thanks for the feedback, I totally agree about the button/hotkey thingg, I'd considered that myself.

I guess, in the ideal world, we'd have a way for the UI to 'grow' with the user comfort level. (I've been a massive fan of http://lunar.lostgarden.com/Mixing_Games_and_Applications.pdf for years).  But in this case, just the hotkey is a great compromise.

Correspondingly, I'm not sure that tabs in the search dialog actually simplify things.  For notepad+, you can see that there's a shed load of functionality (Scintilla2 has a lot of this too, regex replacements etc..), so segregating the UI to keep things simple is important.  For mu, I'm fairly sure we'll never want that level of stuff, so a simple dialog probably is the most straight-forward to understand/maintain.

I did think about two hotkeys, i.e. Ctrl-F to find, and Ctr-H (or Ctrl-Shift-F or...) for replace, but, a) there's no reliable standard for a search-replace hotkey(cross platform), and b) if the functionality is not visible in the interface, then minimizing the number of hidden shortcuts is good.

A compromise option might be to have the find dialog initially simple, with a 'replace' button that adds the extra UI bits:

[image]

alistair-broomhead, 2016-02-24:
I'm not entirely sure what a replace toggle does that a tab does not. You could maybe make the Find label a dropdown, where selecting Replace shows the extra buttons?

carlosperate, 2016-02-24:
I would also like to mention that I personally quite like the idea of keeping the application source code as simple and slim as possible. Having an easy to understand IDE that students can hack around sounds like a great learning opportunity to actually demonstrate how Python can be used in real applications.

NDevox, 2016-02-24:
Alright, with the above in mind my opinion is as follows:

1) If we want to keep things as minimal as possible, let's not hide the replace. what there is now looks good. Maybe add an "(optional)" to the replace label, but even that might be unnecessary.

2) If we really want to be minimal and not add buttons or drop down menus - lets not have it. It just means we have an obfuscated function which people won't find except by accident. And if they trawl the code it would be odd to see it unattached to anything but a hotkey.

This is the kind of thing that would be great to iterate over with user testing!

alistair-broomhead, 2016-02-25:
@NDevox maybe we leave this feature for when we look at #81 - we could always do some form of dynamic UI based on user preferences any how far they've been through onboarding/what other features they've used. 

I definitely don't think we want to build the next PyCharm - as much as I love the IDEA platform, these big lumbering feature-heavy IDEs take years to learn fully, and the functionality you're not using gets in your way in the meantime.

NDevox, 2016-02-26:
@alistair-broomhead I think waiting till we deal with #81 is the right thing to do personally.

Not sure about the comparison to PyCharm :P. Whilst I love IDEA as well, we are a very long way off making something even comparable!

alistair-broomhead, 2016-02-26:
Heh - I was only thinking of IDEA as an example of the extreme other end of the spectrum. Every release there is some cool new feature that I enjoy, but only because I've spent enough time with the platform to have enough familiarity with the established features. Keeping the other end of the spectrum in mind can help to avoid feature-bloat ;)

obrasier, 2017-02-05:
I just wanted to add my +1 to this issue. I think it's a fairly fundamental feature of any kind of text editor. When people copy+paste code to notepad++, and then back again once they have done a find/replace (as my students do) then I think it is something that need to be looked at.

ZanderBrown, 2018-03-23:
So now we are very panel based we could implement this that way

bennuttall, 2018-06-09:
I've been using Mu for a kata recently and this is the main thing missing from it a simple editor IMO. It's not an advanced feature (you'll find it in Notepad, gedit and others), and I pretty much expect it to be in any text editor.

I definitely agree it doesn't need a panel icon, just the shortcut keys ctrl+f and ctrl+h.

Is there a plan to include this for the 1.0 release?

ZanderBrown, 2018-06-09:
No, but I don't think we are against including it in 1.0

ntoll, 2018-06-09:
@bennuttall @ZanderBrown +1 on that. When I used Mu for PyWeek that was the feature I missed (actually it was the only feature I missed). I think, if we can, it should go into 1.0 and I agree with Ben that activating it via the usual hot-key combo is best. In my mind's eye I imagine some sort of modal pop-up rather than a pane (since that's how it usually works AFAICT and we should follow the usual customs so it's familiar to our users when they graduate to a "proper" editor). Does this make sense?

bennuttall, 2018-06-11:
Sounds great - minimal popup modal makes sense :+1:

bennuttall, 2018-06-29:
Awesome, thanks @ntoll. Honestly don't think anyone should replace you with me though...