mu-editor/mu #92: Saving a hex file from Mu

whaleygeek, 2016-03-03:
If I want to save a hex file somewhere other than the micro:bit (e.g. to email it to someone to try), how do I do this? Mu seems to always want to save to an attached micro:bit, and does not give any options to save it elsewhere.

A suggestion could be to CTRL-click the FLASH button, so that it asks you where to save it. This could then also be used if you have more than 1 micro:bit attached, to choose which one to flash to (I currenty have 3 attached for a project I am building!)

alistair-broomhead, 2016-03-04:
This is a good suggestion, but is perhaps best left until we look at #81, to get a holistic idea at to what we want the patterns of interaction to be. The idea of a modifier+click on a tool button is not a bad one, but there may be another idiom that works better for other functionality. 

Given that the main target audience for this is school-age children, we probably want to be more careful than usual about ensuring consistency in our design, and we have the other limiting design goal of keeping the source code as easy to understand as possible, so it can be used as a learning resource by those same children.

MadBashify, 2016-06-06:
i haven't tried this but it could work:
unplug microbit when it says find microbit lead it to a new folder called something like microbit games

whaleygeek, 2017-02-01:
This does work.

closed 2017-02-01 by whaleygeek