mu-editor/mu #93: First time flash of micropython to get REPL not obvious

whaleygeek, 2016-03-03:
To use the REPL, I have to flash something micropython based onto the micro:bit first - the way to do this of course is to press NEW then press FLASH. You can then open the REPL and interact with the micro:bit. However this seems non obvious to new users.

carlosperate, 2016-03-03:
The current [warning message](https://github.com/ntoll/mu/blob/af34dea53dbad427ae97272c08bec8c13bda9dce/mu/logic.py#L174-L178) indicates that MicroPython should have been flashed before REPL will work.

Is the main concern that the message is not clear, or that the most users will not really read it properly (which is a very valid concern)?

whaleygeek, 2017-02-01:
This works. Thanks.

closed 2017-02-01 by whaleygeek