mu-editor/mu #94: REPL text getting damaged on re-edit

whaleygeek, 2016-03-03:
If I type something into the REPL, press return to run it, then up arrow and edit it, and press return again to re-run the edited version - everything past where I press return gets deleted from the REPL window.

e.g.
>>>display.scroll("Hello nicholas")
[up arrow and cursor to the n]
>>>display.scroll("Hello n^icholas")
[change n to N]
>>>display.scroll("Hello N^icholas")
[press return]
>>>display.scroll("Hello N
[I see the above in the REPL window]

whaleygeek, 2016-03-03:
This looks to be the same as #82

ntoll, 2018-06-04:
Just checked... this is fixed in master. Closing.

ntoll, 2018-06-04:
Thanks for the bug report, sorry it has taken so long to reply! :-)

closed 2018-06-04 by ntoll

DavidWhaleMEF, 2018-06-04:
Thanks @ntoll !!