mu-editor/mu #105: Mu has wrong keymap when connecting with tightvncserver on Raspbian Jessie on RPi

jafine, 2016-05-20:
Using realvnc from Mac OS X, to connect to the RPi, the keyboard map is wrong for Mu but fine for everything else e.g. terminal windows, browser etc. It's fine from the main console, so only exhibits this behaviour using tightvncserver. There seems to be a history of this with tightvncserver but has been fixed by setting export XKL_XMODMAP_DISABLE=1 in ~/.vnc/xstartup. In this case it only affects Mu. I tried it on the RPi 3 and RPi 1B models.

carlosperate, 2016-05-20:
I had the same issue in https://github.com/ntoll/mu/issues/61, I've seen this problem with a few debian based distributions on different embedded boards (64bit debian, ubuntu and raspbian), didn't really have too much time to investigate further, so I'm quite glad you found a solution!

jafine, 2016-05-20:
Sorry I meant that although it has that setting in ~/.vnc/xstartup, it is happening on Mu over VNC. The reason I raised this is that it only seem to affect Mu, and other apps are fine. The main difference to other apps I looked at seems to be qt5. After some searching I came across this, which seems to be it: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=814959 . I guess this can be closed if it is a qt5 issue.

ntoll, 2016-05-20:
@jafine thanks for the feedback and I agree that this appears to be a Qt5/Debian issue. Ergo, closing.

closed 2016-05-20 by ntoll

jafine, 2016-05-20:
It actually works fine under x11vnc so it's a tightvnc/Qt5