mu-editor/mu #110: 'missing' msvcr100.dll in Windows 64-bit

hikoel, 2016-06-04:
Running the **mu.exe** under Windows 8.1 64-bit there's an error message saying _"program cannot be started because msvcr100.dll is missing on the computer"_
Solution: there is **Microsoft Visual C++ 2010 Redistributable x86** needed to be installed -- yes: the 32-bit version = x86
DL: [https://www.microsoft.com/en-in/download/details.aspx?id=5555](url)

carlosperate, 2016-06-07:
I believe the dll could be grabbed by PyInstaller to be bundled with the application, but I will need to find out what is the redistributable license for this microsoft libraries area in case it is not permitted and/or requires additional licenses to be included.

It'd be a shame having to list the Visual C++ redistributatable package as a dependency.

hikoel, 2016-06-18:
Speaking for myself I don't mind at all needing to install that "third party pack". With a size of 5 MB it's quite neat and as a windows user I don't hate MS :-D
I asume that you compiled the mu app with Visual Studio, so this is how it came to your "dependency".. so what..

ZanderBrown, 2016-06-18:
Mu is in python so it's not compiled and I've never had this error with python so by process of elimination I'm thinking this is qt's problem

carlosperate, 2016-06-18:
The issue is because we are bundling python inside the mu.exe file in order to have a "one click runnable" application without any additional dependencies. This is specially useful for deploying the application in restricted computers, like you might find in school networks, so it is an important feature to facilitate Mu use in these environments.

Python on windows is compiled using visual studio, which is the reason it links to this specific library. It is possible to include this dll into the bundled Mu executable, but I haven't had the chance to look into the implications of doing this with respect of licenses. If anybody has some time to find this information I'd be quite grateful :)

ntoll, 2016-06-21:
This appears to address redistributing DLLs: https://msdn.microsoft.com/en-us/library/ms235299.aspx

Another alternative is to use a statically compiled version of Python (if such a thing exists). I'm not sure how best to proceed although I know @tjguk has lots of Windows experience, is a Python core developer and may have a wise opinion on how best to resolve this situation.

@tjguk..?

carlosperate, 2016-06-21:
Good news, it sounds like this might have been an issue with PyInstaller not building statically: https://github.com/pyinstaller/pyinstaller/issues/1974

So we could revert back to an older version and update to the latest once it is released (the commit fixing this was pushed a couple of days after 3.2 was out). Alternatively we could git clone and run the latest from the repository, but I don't think it'll make much of a difference to revert back.

The first thing I'll need to do is try to get a clean windows VM without the VS distributables to replicate this issue.

carlosperate, 2016-06-22:
Okay, so I updated the build server script to install PyInstaller 3.1.1 and that seems to fix the issue :)

You can download a version that should work without the redistributable from the normal location: http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/windows/

Direct link to latest version at time of writing: https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/windows/mu-2016-06-22_23_40_41.exe

ntoll, 2016-06-23:
Great stuff!

ntoll, 2016-06-23:
@hikoel does that fix your problem?

hikoel, 2016-07-01:
Yes, so future users will have it easier using the plain _mu.exe_
Likely especially in the 64-bit space, because normally you don't have the _32-bit_ Redistributables on your _64-bit_ system

MichaelToborg, 2016-08-15:
Hello to all,

All information about this DLL error you can find here : ([url]http://www.msvcr100dll.com/(url))

PR #114 (2016-08-23): Specify pyinstaller 3.1.1 to avoid VC++ redistributable dependency.
Tested on a clean Windows 7 32-bit VM. Builds previous to this commit throw the missing dll error, and reverting back to an older version of PyInstaller fixes the issue.

Fixes #110
closed 2016-08-23 by ntoll