mu-editor/mu #112: adding line spaces

damianburrin, 2016-06-19:
When you write a python file in Mu, save, close and reopen the file it adds a blank line between every line

Thanks
DAmian

ZanderBrown, 2016-06-19:
Sometimes it does sometimes it doesn't I'm thinking it's to do with the whole /r/n /n /r nightmare that has plagued every cross-platform editor from the begging of time.

kieldoyle, 2016-06-25:
I've seen this also. I did some testing which i hope might be useful in tracking down the problem.

Start with a simple .py file created outside of mu (line endings in square brackets for clarity):
```
#first comment[CR][LF]
#second comment
```
If i edit the file in mu to delete the line break between the comments and re-add it, then save and close the file i end up with this:
```
#first comment[CR]
[CR][LF]
#second comment
```
When i re-open the file in mu i now have a blank line between the comments and if i save and close again, without making edits, it becomes:
```
#first comment[CR][LF]
[CR][LF]
#second comment
```

carlosperate, 2016-08-06:
Just to point out that this also causes issues with the check functionality, as the temp file created by Mu before it is opened to be analysed by the checker has this extra lines added to it.

closed 2016-08-23 by ntoll