mu-editor/mu #113: Needs more compatibility

MadBashify, 2016-06-20:
I need MU to work with custom python modules as well as a module for the Max7219 LED matrix

inglesp, 2016-06-20:
Hi @MadBashify,

I find the tone of your request highly inappropriate.

I don't know who you are, or how much experience you've got with open source software development, so perhaps you don't understand how these kinds of things work.

Mu is maintained entirely by a small team volunteers, some of whom have poured their heart and soul into the micro:bit project.  You seem to be demanding that they do yet more work without offering to put in any effort yourself.

Please could you consider rephrasing this issue?  Perhaps you could make it a request instead of a demand?  Perhaps you could could thank the team for the work they've done?  Perhaps you could even offer to help out yourself?

Cheers,

Peter.

MadBashify, 2016-06-20:
I'm sorry to come out as rude, I didn't mean it that way I just was in a hurry when I typed it.
I myself am not an advanced programmer; i just do it for fun and so I don't know how to implement things like that myself. Again, I'm sorry for coming off as demanding, i just meant it as a request more than anything. I have edited the post as you may see.
Thanks
-Will

inglesp, 2016-06-20:
Thanks Will -- that's much appreciated.

ntoll, 2016-06-21:
@MadBashify not quite sure what you're asking for. Mu is (currently) only a code editor for the micro:bit. What exactly do you mean when you say "work with custom Python modules"? Perhaps you mean you want MicroPython to work with "custom Python modules". If I understand you correctly, it already does: https://microbit-micropython.readthedocs.io/en/latest/tutorials/storage.html#mainly-main-py

As for the Max7219 LED, a quick Google suggests to me that it should be possible to read the data sheet and plug it into the IO pins for some fun hardware hacking. If you're unsure and/or need help, I suggest you spend some time in the wonder that is the MicroPython forums (available here: http://forum.micropython.org/). It's a very friendly community and if you ask with enough detail/information then people generally send pointers.

Hope this helps..!

ntoll, 2018-06-04:
Closing since this issue appears fixed and/or forgotten.

closed 2018-06-04 by ntoll