mu-editor/mu #119: Fix user's groups on Linux

kzar, 2016-07-19:
I found that I needed to add the current user to the `tty` group before I could use the REPL. The error message displayed would have been pretty confusing to kids. Perhaps if the user doesn't have permission to write to the device we should help them add the current user to the required group(s)?

rhubarbdog, 2017-11-25:
group dialout also works, i've seen group uucp listed for this but haven.t tested it.  If REPL isn't going to work all working groups should be in a dialouge

ntoll, 2018-06-04:
Thanks for the heads up... I'm closing this since:

* This is documented in both the developer docs (mu.rtfd.io) and the new version of the website (coming soon).
* In my experience Linux Mu is not installed by beginners but by (technical) workshop volunteers.
* Mu mentions permissions etc if it can't find the device.

closed 2018-06-04 by ntoll