mu-editor/mu #122: Mu interface on small screens

royletron, 2016-07-24:
Hey Mu team, firstly I want to say excellent job!

I am trying to get Mu running on a Pocketchip. Which I have been successful in, having it run correctly, but I am struggling to get it usable when the ui is so big. The Pocketchip has a 480x272 px screen.

Are there any simple steps to scale the ui down via QT? Or would I need to produce an alternate ui, and can this be done via CSS!? Happy to contribute the work of someone can point me in the right direction!

ZanderBrown, 2016-07-24:
That should be doable it's more how usable the end result would be that concerns me

royletron, 2016-07-25:
Hey @ZanderBrown. As I say, I have got Mu to actually boot and do it's thing, so I am now just concerned with how to refine the UI to get it to work at such a small scale.

carlosperate, 2016-07-25:
How does the window behave in the poketchip? Taking in consideration the low resolution I assume the window is larger than the screen and you have to drag it around to reach all the options ?

Anyway, you can reduce the icon sizes here: [mu/interface.py#L349](https://github.com/ntoll/mu/blob/328684e9646348e138477b92a6987e0aba929d90/mu/interface.py#L349)
and the window minimun size here: [mu/interface.py#L691](https://github.com/ntoll/mu/blob/328684e9646348e138477b92a6987e0aba929d90/mu/interface.py#L691)

royletron, 2016-07-25:
Thanks @carlosperate dragging isn't great! The hope is to get it to all display on one screen, with super small icons as required! I will have a play with the suggestions and see how it comes out.

ZanderBrown, 2016-07-25:
I'm thinking you would probably want to make the window maximised and undecorated as well. I would suggest lowering the padding / margins but with such a small display you would be in danger of clicking 3 things at once!

RogueM, 2016-08-03:
I second the kudos for the good work done so far on mu. I would also like to point out that, while usable, the UI is put of the box a bit too big to fit on the Raspberry Pi official touchscreen (800 x 480 pixels).

jg00dman, 2017-02-07:
In addition to installing latest from git by doing:
`sudo apt-get install python3 python3-setuptools python3-pyqt5 python3-pyqt5.qsci python3-pyqt5.qtserialport python3-serial python3-pep8 pyflakes`
`git clone https://github.com/mu-editor/mu.git`
`cd mu`
`python3 run.py`

If you go into mu directory and edit interface.py edit lines 534 and 535 to 
`       self.setIconSize(QSize(25, 25))`
 `       self.setToolButtonStyle(0)`

also edit line 891 to
`self.setMinimumSize(800, 400)`

It now fits beautifully on a raspberry pi official display.

jg00dman, 2017-02-07:
I´ve written a nice little fix for this which allows the interface to resize dynamically. I like to commit the fix but not sure how? Haven´t ever contributed before but needed this working for students at school earlier today working on raspberry pi.

jg00dman, 2017-02-07:
update - pull request submitted from here: [https://github.com/jg00dman/mu]

ntoll, 2017-06-22:
Fixed in #246

closed 2017-06-22 by ntoll