mu-editor/mu #123: Support more micropython boards

eduvik, 2016-07-28:
This is a neat and simple little editor; I'd like to see more micropython board supported. I've made a fork and am trying to implement support for the ESP8266 - all comms is done via the serial port. 

Are you interested in having multiple board support, or should I go it alone on an ESP8266 fork?

ntoll, 2016-07-29:
+1 I like this a lot since:

* It widens the scope of Mu.
* It points a migration path from the micro:bit to another MicroPython device.
* The more help, the better.

How can I help you with your efforts?

:-)

askvictor, 2016-07-29:
Fantastic! I've already started on my fork, and have it basically working. We'll need to work out switching between the different board and logic sections.

askvictor, 2016-07-29:
Eek, wrong account. I work via @eduvik as well

carlosperate, 2016-07-29:
I'm also interested on this and I'd be happy to help on where I can.

I was just about to raise the same question about switching boards, since some internal state will have to change based on the type of platform targeted we need to indicate/recognise this platform somehow.

The templates branch contains the idea of selecting a "type of project" when creating a new file. But we don't really have a way to identify this type when opening an existing script, unless they have been created in the `python/mu` directory. We could scan the code and check for specific imports?

ZanderBrown, 2016-07-29:
maybe much as we use `# -*- coding: utf8 -*-` to signify utf8 to IDLE & Co we could start files with `# mu-type: bbcmicrobit` to signify that this targets microbit or `# mu-type: pygame` for PyGame ext. Assuming the user doesn't completely empty the file I think this would be an easy & relatively foolproof way to implement it

carlosperate, 2016-07-29:
That was my first though as well, but to be honest the when I started coding in Python I found bangs a bit confusing and magical (not in a good way), even though I had plenty of experience with other languages. I think we should prioritise a coding experience that tries to reduce as much as possible the friction of having to explain boilerplate, and having something like that right away on the first line of a hello world would be against that.

ZanderBrown, 2016-07-29:
i guess we could detect whether or not pygame has been imported but could be tricky to detect different boards

eduvik, 2016-07-31:
I see two approaches here: 

1. A config option for the entire IDE. This will likely suffice for most use cases, as most users will probably only use a single type of board. This could be implemented on the button bar, showing what the currently chosen board is to make that visible.
1. A config option for each project. I'm aware that there isn't really the concept of a project in mu at this point, or rather there is only the one project in ~/python. However, when people come to re-use code, or want to give a friend their project to flash, then I think that might cause problems. One possibility might be to keep a .mu-project-config file/directory inside the project directory that holds this info (nice that it keeps it together with the project, so copying the directory carries this info). Another would be to keep this in the global config; each project has an entry with project-specific info (nice in that it keeps mysterious files away from the user).

I'd be inclined to include .mu-project-config in the project directory.

eduvik, 2016-07-31:
It is also worth thinking about how boards will differ from the programmer's point of view.  

On one hand, different boards will have different GPIO ports and different IO built it, so a program for one board probably won't run on a different board, so you'd want to specify what board the program target somewhere. 

On the other hand, we could let the programmer work that out (assuming beginners will mostly be using only a single type of board), and give them nothing other than choosing the serial port/mechanism to send files to the device,

eduvik, 2016-07-31:
I've got basic functionality working with the ESP8266 at https://github.com/eduvik/mu/commits/master (sorry about committing to master; I want a quick proof of concept, and will set up branches now that it seems to be feasible to go forward). 

I don't have any other types of board to test other than a  few types of ESPs, but this should work on any device that communicates exclusively over the serial port.

eduvik, 2016-08-07:
I've abstracted the device into a seperate module, and have REPL working with serial-only ESP8266, webREPL ESP8266, and have untested  code for the microbit (I've copied the code from the existing codebase, so it should work identically, but without a device to test on I can't make any guarantees). File transfers are not yet supported, but that's next on the agenda.

Type of board and connection parameters are set in a file config.py.

https://github.com/eduvik/mu/tree/feature/multi-board

carlosperate, 2016-08-12:
If you have a branch that could be submitted as a PR, even if it's not ready for merge, it would make it quite easy to open a discussion and get it ready for merge :)

eduvik, 2016-08-12:
Here you go: https://github.com/ntoll/mu/pull/135

ntoll, 2018-06-04:
Hey hey @eduvik -- I bet you can't tell I'm doing some GitHub gardening today..? ;-)

The new architecture for Mu (modes etc) should make it *really easy* to add new boards. See https://mu.readthedocs.io/en/latest/modes.html#create-a-new-mode for the high level view of how this works. Perhaps post 1.0, care to submit an ESP8266 one..? :-)

closed 2018-06-04 by ntoll

dybber, 2018-06-20:
I've started working on a new mode for ESP8266 (and ESP32). The REPL, flashing and filetransfers seems to work now, however there are still af few TODOs and some debugging left. I'm using the Python libraries from `ampy` for datatransfers https://github.com/adafruit/ampy

https://github.com/dybber/mu/tree/esp8266-mode

Cheers,
Martin

ntoll, 2018-06-20:
Great stuff and many thanks for your work! :-) 

Just a quick heads up on Mu timelines: 

* Beta.16 will arrive *today*. 
* I expect beta.17 to arrive about a fortnight afterwards. 
* A release candidate a week after that. 
* Assuming no problems, a 1.0 final a few days later. 

I'm not sure ESP* mode will get in with enough time for comprehensive testing. However, I intend to release 1.1 soon after 1.0 for the sole purpose of adding a couple of new modes of my own. Perhaps the ESP* mode can be part of that release..?

Thoughts..? (I'm flexible about all of this.)

dybber, 2018-06-20:
Hi Nicholas!

Thanks for a quick reply :) My hope is to use Mu as part of a beginner programming course I'm teaching in august 13th-24th. I think you are right it will take some time to test this on all platforms. 
If it could be part of a 1.1 beta release during july, I could let my students use the 1.1. beta. That would be nice :)

Furthermore, I think there is a longer term potential to merge all these MicroPython modes into a single mode (CircuitPython, Microbit, ESP8266), and depending on VID/PID select the correct APIs to use.

Best, Martin

cauebs, 2018-08-23:
Any progress on this mode? I, too, was interested in using Mu to teach MicroPython. I might be able to contribute, if necessary :)

dybber, 2018-08-23:
Hi @cauebs. Definitely! I'm currently using it to teach a course of 50 students. Try it out from this branch: https://github.com/dybber/mu/tree/esp-mode

dybber, 2018-08-23:
Next steps will be writing a test suite, if you are interested you are welcome to start working on it. Otherwise, I will try to find some time for it in september/october.

jedie, 2018-10-01:
@dybber Any news about your fork?

I tried it out. It's useable... However, I think thonny microPython support is more advanced, see: https://bitbucket.org/plas/thonny/wiki/MicroPython

dybber, 2018-10-04:
Hi @jedie. My september has been fully occupied by other activities, but hopefully I will soon be able to write a test suite. I didn't know about Thonny. What features do you find missing? The feature for executing all code in a pane, without transferring it as a file it seems nice.

Here are the main annoyances from my perspective:
 - The file manager is really bad, and could use improvement, especially support for navigating between folders and making buttons for transfering files instead of drag-and-drop
 - We should perhaps just use the build in Open File/Save File dialogs to access the PC filesystem, as currently you are not able to transfer files from outside mu_code
 - When the REPL is open it would be nice with a "Reboot" button that sends Ctrl+D
(- Being able to modify the location of the mu_code folder from a preferences dialog, but is a more general comment for Mu)

Other than that it has worked nicely for my course, and will continue using it.

yurikleb, 2018-10-16:
Hey @dybber I just ran some test using your [branch](https://github.com/dybber/mu/tree/esp-mode) on Ubuntu 18 with the [M5STACK](http://m5stack.com/) and it works quite well.
Great Work!

I have one question, on the M5STACK all the files need to be placed in a /flash/ folder.
I can see the flash folder in the file manager, but I'm unable to see the files inside it.
Is this a feature which i not implemented yet? or is there some setting I could modify?
Thanks! 

Basically the M5STACK has quite good [Web-IDE](http://io.m5stack.com) but I'm working on a project in which the WiFi AP has no access to the Web, so MU seems like a great solution!

Thanks again for the great work!

dybber, 2018-10-16:
Hi @yurikleb 

I've been using the file manager pane from the micro:bit mode which currently is very limited. It is definitely one of the more annoying things right now, which I'd like find a better solution for.

I didn't know about M5Stack before, and will probably not be able to acquire the hardware myself, but would be nice if we could support it.

yurikleb, 2018-10-16:
Hey @dybber thanks for the quick response!
The M5STACK is basically an ESP32 nicely packaged with a screen, battery, 
IP5306 battery management, some buttons, MicroSD slot, Speaker, Mic and few other handy features, so as long as all works well with an ESP32 the M5STACK should also be supported.

Pillar1989, 2019-04-30:
I really hate to see @yurikleb  who comes to post ads in such a professional technology discussion area

yurikleb, 2019-05-02:
Hey @Pillar1989 ,
I'm sorry I didn't mean to make it commercial. I have no affiliation with the company behind M5Stack.
I do think it's useful to share information about which hardware is compatible, as the universe of micro python boards is growing.

I think mu-editor is an awesome project, and there is nothing wrong about sharing information regarding which hardware it was tested with and plays well / has issues with.