mu-editor/mu #125: Error attempting to run Linux binary

rotacoder, 2016-07-29:
I've downloaded the Linux binary for my Debian system but after making it executable I get the error message "cannot execute binary file". I'm assuming this is because my system is 32 bit and the binary is for 64 bit. If this is the case, will there be a 32 bit version of the editor in the future?

Thanks!

carlosperate, 2016-07-29:
You can install it on 32 bits using the deb file, as all it does is install all the listed dependencies and uncompress the source code on the right place and run the application using Python.
At the moment there is no plans to start building 32bit linux binaries, as not many free CI servers offer 32 bit OS images.

The deb hosted by the raspberry pi repository can be found in http://archive.raspberrypi.org/debian/pool/main/m/mu/ but it contains an old version of Mu.
The deb files automatically generated by the build servers can be found in: http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/raspberry_pi/
At the time of writing the latest version is https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/raspberry_pi/mu-2016-07-17_19_57_42.deb
These deb files have not really been tested properly, so there is a small chance they might not work, but it would be great if you could try it and report back :)

rotacoder, 2016-07-29:
Thanks Carlos. I downloaded the deb but got some errors about missing modules when trying to run mu. The first was for "pyflakes" which I installed using pip3. The second was for "pycodestyle", which again installed ok. Attempting to run mu then resulted in another ImportError, this time for "serial". However, when trying to install I got this error:

~$ sudo pip3 install serial
Downloading/unpacking serial
  Could not find any downloads that satisfy the requirement serial
Cleaning up...
No distributions at all found for serial
Storing debug log for failure in /root/.pip/pip.log

Not sure where to go from here...

carlosperate, 2016-07-29:
>The first was for "pyflakes" which I installed using pip3. The second was for "pycodestyle"...

Yeah, that's exactly what I though might happen, I've been looking at how to include these dependencies since my last reply, as these are usually installed through pip and the deb package should only fetch new stuff from aptitutde.

I'm working on it and should have a new deb for you very soon.

rotacoder, 2016-07-29:
Ok thanks Carlos. I'll check back periodically.

carlosperate, 2016-07-29:
Try https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/raspberry_pi/mu-2016-07-29_23_01_16.deb and see if that works correctly.

rotacoder, 2016-07-30:
Hi Carlos,
Success! 
First I got a bunch of dependency errors but doing 'apt-get -f install' fixed them. I don't actually have a micro:bit yet but I'll let you know if there are any issues when I get it.

Thanks very much for your prompt attention on this!

ntoll, 2016-07-30:
@carlosperate great work! :-)

carlosperate, 2016-07-31:
I've created a PR to master to include these changes, but we still need to check if it'll work on a raspberry pi: https://github.com/ntoll/mu/pull/127

carlosperate, 2016-09-26:
I think we can close this as #127 has been merged and #143 has been opened to cover any additional work.

closed 2016-09-26 by carlosperate