mu-editor/mu #126: Mu creates directories in my home

deshipu, 2016-07-30:
Every time I start Mu, it creates a directory called "python" in my home directory. Not only is this extremely annoying, but it also took me some time to figure out which application is responsible for this, as the name is completely unrelated to Mu.

If it needs a configuration directory, I would suggest using ``~/.config/mu`` by default (or even better, follow the xdg standard for configuration directories).

ntoll, 2016-07-30:
Hi @deshipu!

Mu *does* use the standard location for configuration. The `python` directory is for your source code. The feedback I got from Carrie Anne was that kids (and teachers) wanted a single standard place for their code on all the computers they used. Mu, by default, checks the expected `~/python` directory for loading and saving. 

How about I add a configuration option for default source code directory..? If you're `1337` enough to get annoyed by the `python` directory you'd certainly be `1337` enough to read the docs and update the settings in the very simple JSON file that Mu uses. ;-)

deshipu, 2016-07-30:
Can you also make it so that if it's set to "", no directories are created at all?

deshipu, 2016-07-30:
By the way, not sure if "python" is a good default name, you can get conflicts with other python-related tools that had a similar idea...

ntoll, 2016-07-30:
Upon reflection it should be something like `mu_code` - it names the application *and* tells you what's in it.

deshipu, 2016-07-30:
That sounds great, thank you. Sorry for complaining, but I figured that I will tell you, and you can ignore it if there are more pressing things to work on.

carlosperate, 2016-07-30:
I was thinking about this as well a few days ago when reading https://github.com/ntoll/mu/issues/86.
Having the option in the config file would be a good alternative to be able to configure this option if required, but not having to worry about it.

ZanderBrown, 2016-07-30:
as templates and project types are being added a `Mu Projects` or `Mu Workspace` folder would probably be better than a `python` folder, although we do need to be careful not to form a directory structure of too much complexity the 'stick everything in one folder' approach may not be the best

tibs, 2016-09-19:
I'm working on changing the name to mu_code, which seems simple but fairly self-explanatory.

deshipu, 2016-09-19:
I would still love to not have it at the top level of my home directory.

carlosperate, 2016-09-19:
For that we will add a new entry to the settings file to select the project folder location. The change @tibs is mentioning will only modify the mane of the folder itself.

ntoll, 2016-09-22:
@deshipu Having it in the top level of your home directory is precisely where teachers and students have told us they want to find it! ;-)

Of course, you should be able to override it as per #86.

closed 2016-10-15 by ntoll

deshipu, 2016-10-16:
I don't think #156 fixes the original issue reported here.

PR #156 (2016-10-15): Fix #126 Change name of default Python directory
Change the name of the default Python (save/load) directory from ~/python to ~/mu_code, which fixes #126

Also:

- Add a changelog, CHANGES.rst
- Update the development documentation (in README.rst and docs/development.rst) to explain how to install PyQt5 and QScintilla on Mac OS
- Add me to the AUTHORS file