mu-editor/mu #128: files/REPL dialog box needs to be a modal dialog

peterhinch, 2016-08-03:
The dialog box which pops up if you click on files when the REPL is open (and vice versa) does not behave like a proper modal dialog: if you click on the main window it disappears behind the main window and the application appears to be unresponsive. For a beginner-focussed app this is confusing: the dialog box should remain in the foreground.

peterhinch, 2016-08-13:
The problem is that a modal dialog needs a parent window otherwise it is "owned" by Qt rather thn by a specific window, So here's how to fix it (docstrings omitted for clarity)
```python
    def show_message(self, message, information=None, icon=None, parent=None):
        message_box = QMessageBox(parent)
```
and
```python
    def show_confirmation(self, message, information=None, icon=None, parent=None):
        message_box = QMessageBox(parent)
```
Then in each call provide
```python
            self._view.show_message(message, information, parent = self._view)
```

ntoll, 2016-08-14:
Hi @peterhinch 
Great stuff!
Care to turn this into a PR? Then your contribution will be immortalised forever! :-)

peterhinch, 2016-08-15:
I would if I weren't well out of my depth with GitHub. I'm actually working on a fork targeted at the ESP8266 running MicroPython. My [fork](https://github.com/peterhinch/mu) is forked from [eduvik's](https://github.com/eduvik/mu) which is in turn forked from your code. While I can raise PR's against eduvik I'm lost as to how to do it against your code.

carlosperate, 2016-08-15:
When you create a pull request from your repository page it will most likely show you a comparison between your selected branch and eduvik's master. If you click on that drop down menu you should be able to select any other repository, including ntoll's mu master.

peterhinch, 2016-08-16:
OK, I've raised a PR. I hope I've got it right - I find Python a whole lot easier than Git ;-)

ZanderBrown, 2016-08-16:
travis isn't happy but otherwise everything appears to be in order. It does help when you reference the pull (#138) and you appear to have renamed `find_microbit()` to `find_upython_device()` and your pull doesn't reference this

peterhinch, 2016-08-17:
This came from eduvik's fork. I clearly failed to disentangle my changes from his.

ZanderBrown, 2016-08-17:
ah. Well would you mind if I was to setup a pull containing only your changes?

peterhinch, 2016-08-18:
No problem.

carlosperate, 2016-09-20:
This issue has been fixed in #141. Thanks for the contributions!

closed 2016-09-20 by carlosperate