mu-editor/mu #129: Libraryify

ZanderBrown, 2016-08-04:
At present Mu is one big mass of code. A very well documented mass but a mass it is.

It's also VERY dependent on Qt.

So my thought is:

Move most of the functionality into a 'mulib' (imaginative I know :smile: ) and reduce the dependency on Qt (especially the likes of QSerialPort) and achieve separation of logic and interface. I know this is already  largely done but I feel I could be improved.

Doing this should make Mu easier the maintain as well as opening up the possibility of extensions. At the moment any new features (say Micro:Bit file browser) get backed into Mu itself. If it was an 'add-on' the feature could be easily loaded/unloaded as appropriate because a Micro:Bit file viewer isn't relevant when editing a PyGame project.

Feel free to ignore me, just making a suggestion for a project I like (even if it does use Qt 😄 )

ntoll, 2016-08-05:
Totally agree with the sentiment. :-)

All the logic should be in, er, `logic.py` and QT stuff in `interface.py` with `app.py` as the bridge.

That there is leakage between the two is problematic and something I want to address. You're right to point out the dependency on the Qt serial library (it's actually why I used PySerial for the most recent file-system based work). I think we're mostly in good shape, but this is for nothing if there isn't complete separation of concerns.

I'm about to go on holiday for a week, but it's certainly something to address. If you're around for PyCon UK it's certainly something for the sprints. :-)

ZanderBrown, 2016-08-05:
Unfortunately I won't be there but I may still have a tinker

ntoll, 2016-08-05:
Great stuff..! As always, constructive contributions most welcome! :-)

ZanderBrown, 2016-08-23:
I haven't forgotten about this but I've been busy with this & that lately. Less to do now so hopefully I'll be able to put something together.

ntoll, 2016-08-23:
Ditto - holidays and family commitments have eaten up my time recently. @ZanderBrown will you be at PyCon UK in a few weeks, a bunch of us want to sprint on this on the Monday sprints day (http://pyconuk.org/).

ZanderBrown, 2016-08-23:
@ntoll I wish I could be there! Unfortunately, I'm unavoidably otherwise occupied during those weeks.

ntoll, 2016-08-25:
@ZanderBrown I see you're in Stevenage. If you're ever in London let me know. Would be good to meet up and chat these things through. Brains in the same room always wins over email and GitHub comments. ;-)

ZanderBrown, 2018-03-10:
I think with V1 this is essentially done

closed 2018-03-10 by ZanderBrown