mu-editor/mu #131: Replace references to "bbc:microbit" with micropython device

askvictor, 2016-08-07:
There are a lot of references to bbc:microbit in the code and strings. If this is going to be a more general editor for micropython, these should be replaced with more generic names. In-code references should just be to 'device' or something similar. End-user string references could either be generic (micropython device), or dynamic, based on the device as determined by the settings (either global or project) - will need a settings structure first: https://github.com/ntoll/mu/issues/123#issuecomment-236405732

ZanderBrown, 2018-04-07:
With V1 and its modes this is probably defunct

ZanderBrown, 2018-05-12:
@ntoll

ntoll, 2018-05-12:
Yes... closing. Thanks for highlighting this @askvictor and @ZanderBrown

closed 2018-05-12 by ntoll