mu-editor/mu #132: REPL-based debugger

eduvik, 2016-08-12:
For lack of a debugger in micropython itself, how about an option to 'debug' a program, that runs the current code file line-by-line via the REPL?

eduvik, 2016-08-12:
Proof-of-concept implemented here: https://github.com/eduvik/mu/tree/feature/repl-debugger

carlosperate, 2016-08-12:
How does that deal with loops and other types of jumps?

eduvik, 2016-08-12:
Rats; hadn't got that far. Might be a problem.

eduvik, 2016-08-13:
Hmm; don't think this will be possible unless REPL has a mode where, when in a loop (or branch), it steps through each line. Which doesn't currently exist.

carlosperate, 2016-08-14:
I believe the kl26 micro has debugging capabilities for the nrf51. We can always revisit this feature once that area has been explored a little bit more.

ntoll, 2018-06-04:
Given the problem with loops... I guess we should close this now. :-/

It would have been such a cool feature. Thank you @eduvik!

closed 2018-06-04 by ntoll