mu-editor/mu #137: RFC User interface simplification

peterhinch, 2016-08-14:
For an app designed for children the dialog box about USB interfaces is confusing and arguably unnecessary.

An alternative is for the UI to behave as follows. If the files toggle is pressed when the REPL is present, the REPL disappears. On a second press, the files appear. Similar behaviour occurs when the REPL toggle is pressed with the files present. Perhaps others could try this and comment.

logic.py
```python
    def toggle_fs(self):
        """
        If the file system navigator is active enable it. Otherwise hide it.
        If the REPL is active, display a message.
        """
        if self.repl is not None:
            self.remove_repl()
        else:
            if self.fs is None:
                self.add_fs()
            else:
                self.remove_fs()

    def toggle_repl(self):
        """
        If the REPL is active, close it; otherwise open the REPL.
        """
        if self.fs is not None:
            self.remove_fs()
        else:
            if self.repl is None:
                self.add_repl()
            else:
                self.remove_repl()
```

ZanderBrown, 2018-04-07:
The Python mode's debugger does provide a precedent for disabling buttons and the current message isn't massively friendly but your left with the "why has that one gone gray" and it may not be obvious you need to close one pane to get the other

ntoll, 2018-04-07:
So certain buttons (including files and repl) are enabled / disabled according to the current state. I'll mark this as closed since it solves the problem outlined by @peterhinch -- i.e. you cannot use files *AND* REPL at the same time.

closed 2018-04-07 by ntoll