mu-editor/mu #142: Track pycodestyle debian package release

carlosperate, 2016-08-25:
The deb package for Mu needs to install all dependencies using the package manager. The old module `pep8` has been renamed to `pycodestyle` not long ago, but a package with the new name has not yet been published.

As part PR #127 we've introduced a conditional import to fall back to the old name in [/mu/logic.py#L32-L36](https://github.com/ntoll/mu/blob/b3502080059ec93048fb11d183cbb784ca7fcc43/mu/logic.py#L32-L36) :

```python
try:
    from pycodestyle import StyleGuide, Checker
except ImportError:
    from pep8 import StyleGuide, Checker
```

We should track the release of the renamed package and correct this import once it is out. More info can be found in https://osdir.com/ml/debian-python/2016-06/msg00041.html .

ntoll, 2016-08-26:
+1

I also notice that the try/except breaks 100% coverage (absolutely not a problem). Would be good to move things over when upstream Debian catches up.

ntoll, 2018-06-04:
Hmmm...

This appears to have been backported to Jessie (https://tracker.debian.org/pkg/pycodestyle). However, I don't see this in any of the Debian related stuff in our repository.

carlosperate, 2018-06-04:
I had this old branch as well lingering, so submited this as a PR since we should be fine to remove it.

carlosperate, 2018-06-13:
#477 has been merged, so closing this one, thanks Nicholas!

closed 2018-06-13 by carlosperate