mu-editor/mu #143: Deb packaging: investigate "interim log entries"

carlosperate, 2016-08-25:
The current raspberry pi CI server that generates PyInstaller builds also generates deb packages and uploads them to: http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/raspberry_pi/

The package version is collected from the history log entries. As it currently stands, these CI deb packages clone the repository, add a new entry to the log, but do not commit the changes (as it does a build for each commit to this repository). It looks like without committing, these log entries are ignore, and so it generates all packages with the same version number.

If a user has already installed a proper deb release, and these CI generated packages have the same version number, and it looks like any new dependencies are not updated on `sudo apt-get -f install`. Just to clarify, the code changes included in the package itself seem to be correctly installed replacing the old code.

We need to investigate a solution to this. On PR #127 we've discussed the possibility to manually commit a new "interim" entry to the log every time the dependencies are changed, so that it upversions the package and the package manager can then correctly detect these changes.

carlosperate, 2018-03-17:
The raspberry pi CI serve has been retired, as it is now in a drawer since I moved houses (but can be revived if needed).
Since we are not building deb files on CI, and they will be manually generated for submission to their respective repositories we don't really have this issue any more and can close this ticket.

closed 2018-03-17 by carlosperate

ntoll, 2018-03-20:
RIP :strawberry: :cake:  :skull: