mu-editor/mu #146: Microbit throws error on OSX

jessingrass, 2016-09-10:
Repro steps:

1. Attach new microbit
2. download latest build (2016-08-25) of mu
3. open build
4. print('Hello, world!)
5. Flash to microbit

Expected Behavior: Hello, world!  flashes to the microbit
Actual Behavior: microbit shows as "discharged" and does not display the code
[image]

ntoll, 2016-09-11:
Hi @jessingrass, thanks for the feedback.

This is expected behaviour but Mu needs to signal this so people don't feel worried that something has gone wrong.

Essentially, what's happening is that the micro:bit is designed to restart once the process of flashing the device has completed. As I'm sure you know, restarting looks like unplugging and plugging in again to the "host" computer to which the device is attached. If that computer is running OSX you get the warning you see in the screenshot you attached.

Any suggestions for how to tell people this is expected..?

jessingrass, 2016-09-12:
Will work on a suggestion for wording. The bigger problem I have is that the "Hello, world!" message is NOT scrolling to the microbit. The orange light flashes indicating that something is happening, but then nothing happens on the LED screen.

ntoll, 2016-09-12:
Can you connect to the device with the REPL in Mu?

ntoll, 2016-09-12:
I see the problem... you're shouldn't `print("Hello, world!")` but `display.scroll('Hello, World!')` ;-)

See: https://microbit-micropython.readthedocs.io/en/latest/tutorials/hello.html

jessingrass, 2016-09-12:
I'm an idiot. Thanks =)

jessingrass, 2016-09-12:
As far as the messaging, maybe just tell people 
"OSX Users: if you get a 'disk not ejected properly' message, don't panic! Simply do x, y, z..."    or
"OSX Users: if you get a 'disk not ejected properly' message, don't panic! This is how mu interacts with OSX..."

ntoll, 2018-06-04:
Closing since this issue was resolved.

closed 2018-06-04 by ntoll