mu-editor/mu #147: Check button on windows exe doesn't do anything yet

dannystaple, 2016-09-19:
Using some sample code:
```
from microbit import *

# Write your code here :-)
display.
print('missing quote)
```

Pressing the check button pops up nothing. Mu does show the missing quote as an error by default.

carlosperate, 2016-09-19:
This might be related to https://github.com/mu-editor/mu/issues/117. Was this the version downloaded from codewit.mu ? Could you try this exe as well? (latest build) https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/windows/mu-2016-08-25_20_51_43.exe

dannystaple, 2016-09-19:
That worked.

carlosperate, 2016-09-19:
Great. We need to make sure the patched executable goes to the website.

ntoll, 2016-09-22:
I've updated the website. Apologies for the delay on my part.... post PyCon UK work got in the way. Only just catching up on PRs. :-)

closed 2016-09-22 by ntoll